micasense/imageprocessing #1: Exif Tags for radiometric calibration missing

dud3r, 2018-04-10:
Hi
I'd like to try out your workflow. Unfortunately my Exif tags do not include `'XMP:RadiometricCalibration'` nor `'XMP:VignettingCenter'` nor `'XMP:VignettingPolynomial'.` Firmware is v1.5.30

Are those information available on every MicaSense camera or do I have to purchase them separately?

By the way I have data from a professional radiometric calibration in the lab (w/ Integrating sphere) for different sets of exposure times & illumination conditions. So i would be glad to transform our own data into a1, a2, a3 and V, x, y respectively if this is possible.

Any help is much appreciated

maykef, 2018-04-10:
Hi, 
If your exif tags do not include these values, it means you have an old firmware in your camera. I had to add them manually for an old set of images I took some time ago.  The process is a bit cumbersome so, I recommend you to update your camera firmware. You don't have to purchase anything else, as soon as you update your camera, it will have those values included.

dud3r, 2018-04-10:
Thank you for the fast answer, I will try to update the firmware and report back if it worked.

dud3r, 2018-04-10:
And there we have it. After updating the firmware form 1.5.30 to 3.3.0 I can read all the missing Exif information. Thanks a lot. Continuing with the tutorial...

closed 2018-04-10 by dud3r