micasense/imageprocessing #3: Correction of lens distortion not working for RedEdge-M

hkristen, 2018-04-12:
I am using the whole camera calibration workflow, as shown in the tutorial, for a RedEdge camera with firmware v2.1.3, without any problems. 

But when I apply the same workflow for images from the RedEdge-M camera (firmware v3.3.0), the  lens distortion correction produces a very weird result (see below).

[image]

I figured, that there must be a problem in the EXIF tags that are used for the calibration. Indeed there is a interesting difference in the tag "Perspective Focal Length":
RedEdge-M: _Perspective Focal Length: 5.4418739708109527_
RedEdge: _Perspective Focal Length : 1459.7272592972727_

All other tags are in a similar value range (see below exif files)
[rededge_exif.txt](https://github.com/micasense/imageprocessing/files/1903792/rededge_exif.txt)
[rededge_m_exif.txt](https://github.com/micasense/imageprocessing/files/1903793/rededge_m_exif.txt)

Here is the RedEdge-M image I used, to reproduce this behavior:
https://goo.gl/pPKkRm

poynting, 2018-04-13:
Hi @hkristen.  Good catch. The units of this tag were changed in v3+ due to compatibility issues with some external software.  It's necessary in v3+ to check for the "Perspective Focal Length Units" field.  If it is not present, the provided focal length is in pixels. If it is present, the provided focal length is in the units provided, which will be mm.  Conversion can be done via the below function:

```
def focal_length_mm(meta):
        units = meta.get_item('XMP:PerspectiveFocalLengthUnits')
        focal_length_mm = 0.0
        if units == 'mm':
            focal_length_mm = float(meta.get_item('XMP:PerspectiveFocalLength'))
        else:
            focal_length_px = float(meta.get_item('XMP:PerspectiveFocalLength'))
            px_per_mm = self.focal_plane_resolution_um()[0]
            focal_length_mm = focal_length_px / px_per_mm
        return focal_length_mm
```

Then you can update 

```
fx = fy = float(meta.get_item('XMP:PerspectiveFocalLength'))
```
to 
```
fx = fy = focal_length_mm() * FocalPlaneXResolution
```

in the utils:correct_lens_distortion method, which should work for both cases.  I don't have an easy way to test this at the moment for both cases, but if you want to try this out and submit a pull request I'd be happy to review it!

poynting, 2018-04-17:
Hi @hkristen that's correct, I was adapting this code from V2 of the tutorials which I'm trying to finish up at this time which has those functions.  There is an EXIF tag that provides focal plane resolution in px/mm. Here are the inter-dependent functions:

```python
def focal_plane_resolution_px_per_mm(meta):
        fp_x_resolution = float(meta.get_item('EXIF:FocalPlaneXResolution'))
        fp_y_resolution = float(meta.get_item('EXIF:FocalPlaneYResolution'))
        return fp_x_resolution, fp_y_resolution

def focal_length_mm(meta):
    units = meta.get_item('XMP:PerspectiveFocalLengthUnits')
    focal_length_mm = 0.0
    if units == 'mm':
        focal_length_mm = float(meta.get_item('XMP:PerspectiveFocalLength'))
    else:
        focal_length_px = float(meta.get_item('XMP:PerspectiveFocalLength'))
        focal_length_mm = focal_length_px / focal_plane_resolution_px_per_mm()[0]
    return focal_length_mm
```

poynting, 2018-04-18:
PR merged.  Thanks @hkristen!

closed 2018-04-18 by poynting

andbrs, 2019-03-05:
hello I've been trying to fix the problem since I use software version 4.2.2 and I get a similar error. I tried to copy the inter-dependent functions in metadata.py but it still gives me a problem. You could explain where these inter-dependent functions should be added or which is the process to fix the error. Thank you very much for these excellent libraries

poynting, 2019-03-05:
@andbrs the issue described here was fixed and merged almost a year ago; if you're having a problem with the current version of the library code, please open a new issue and include more information about the issue you're seeing.

andbrs, 2019-03-05:
Sorry, I already wrote a new post