micasense/imageprocessing #4: Image Processing Tutorial #1 does not deal with metadata in Pictures

maykef, 2018-04-12:
Hi, 

The last line in your code does not save images as TIFFS, nor preserves metadata. 
The consequence is that uploading these images to Pix4D or Agisoft is useless, since all the metadata is lost and, therefore, no mosaic of the band can be created by the programs.

Please advise on using a specific library that preserves Metadata, or a method to copy and paste metadata in last images.

poynting, 2018-04-13:
Hi @maykef, if your intent is to process with Agisoft or Pix4D you have a couple of options:

1. Use the original files. Both of those software packages support RedEdge and implement the radiometric corrections internally.  You don't need to pre-process the files.
2. Use this tutorial as a reference and add your own code to save the files out to tiff using other libraries, then write the required metadata into the output files using exiftool. This can be challenging as most python image libraries don't properly support multi-page tiffs.  You'll have to then determine which metadata is required for processing by your chosen software.  

I'm closing this because the requested functionality is out of the scope of the tutorial.

We'll also be updating the tutorials soon and will provide extended functionality, including an example on saving data out to tiff files using gdal. We don't currently have writing of metadata out implemented, but we'd love to consider a pull request if someone wants to add that.  I would just suggest waiting until the next update to do any work toward that end.

closed 2018-04-13 by poynting

maykef, 2018-04-13:
Understood Justin,

Thanks for this.

dud3r, 2018-04-20:
Ok, one last comment on keeping metadata in the images. I haven't had any luck with Python libraries (pyexiftool, py3exiv2). But running exiftool from the command line has given me success. Photoscan does recognize my radiance calibrated images as MicaSense multispectral bands. The important thing is to copy the XMP tags blockwise (`-xmp`). If you just use exiftool the normal way and try to copy all tags from one file to another (`-all`) they will get "translated" or rearranged internally somehow. So the final command for transfering all metadata from MicaSense raw images to altered images would be:

`exiftool -tagsfromfile IMG_0000_1.tif -xmp Pile/Radiance/IMG_0000_1_RAD.tif`
`exiftool -tagsfromfile IMG_0000_2.tif -xmp Pile/Radiance/IMG_0000_2_RAD.tif`

... and so on.
Cheers.