micasense/imageprocessing #9: How much time is needed for aligning images that are provided in repository ?

Khumayun, 2018-06-14:
How much time is needed for aligning images that are provided in repository ?
I am following step by step workflow to align images, but when it comes to 

> 'Alinging images. Depending on settings this can take from a few seconds to many minutes'

 there appears 4 same windows as from `capture.plot_undistorted_reflectance(panel_irradiance)`. I tried waiting as long as I could, but nothing happened. Please, provide with appropriate solution.

P.S. Can`t upload screenshot, it says `something went really wrong, and we can`t process that file,`

maykef, 2018-06-14:
This will pretty much depend on your system's configuration.
Can you elaborate a bit more on that? What platform are you using, Python version, Opencv installation, etc?
Also, most of these tutorials are informative, is not meant to be a 'support' type page/service. Bear that in mind when open new issues.

Khumayun, 2018-06-14:
Processor Intel® Core™ i3-3110M CPU @ 2.40GHz × 4 
Graphics   Intel® Ivybridge Mobile
OS type    64-bit
OS            ubuntu 16.04 LTS
Python      3.5.2 (default, Nov 23 2017, 16:37:01) 
[GCC 5.4.0 20160609] on linux
[requirements.txt](https://github.com/micasense/imageprocessing/files/2101958/reqs.txt)

Ok, I will take into an account. But those multiple windows occurring at the same time, make me confused.

poynting, 2018-06-14:
Hi @Khumayun,

Using the code and settings in `Alignment.ipynb`, it finishes on my core i7 in about 15 seconds.

At one point during development, I had some troubles with a similar problem. The 'fix' on my system was to add the line
```
multiprocessing.set_start_method('spawn')
```
to [imageutills.py align_capture method](https://github.com/micasense/imageprocessing/blob/master/micasense/imageutils.py#L113). This fixed the issue for me.  The symptom I could see in the debugger was that the `cv2.findTransforECC` call would never complete, and it was clear from CPU usage that it was not doing anything.

If this isn't working, one simple solution could be to just unwrap the `enumerate(pool.imap_unordered()` call into a series of calls, to remove the multi-processing component entirely. This approach worked for me in all cases, but is a little slower as it doesn't use all CPUs efficiently.

closed 2018-06-18 by Khumayun