micasense/imageprocessing #10: Image align - Iteration

filmfer, 2018-07-11:
Hi I'm using the Micasense Lib in github to create a script able to process and get a stack tif file for every single file token from a Micasense Rededge. I used you sample code and sample data and added a simple iteration over the file names to process every file in folder. But i get this error:
"---------------------------------------------------------------------------
RemoteTraceback Traceback (most recent call last)
RemoteTraceback: 
"""
Traceback (most recent call last):
File "C:\Anaconda3\envs\micasense_pnnl\lib\multiprocessing\pool.py", line 119, in worker
result = (True, func(*args, **kwds))
File "c:\micasense\imageprocessing-master\micasense\imageutils.py", line 85, in align
criteria)
cv2.error: OpenCV(3.4.1) C:\bld\opencv_1520732670222\work\opencv-3.4.1\modules\video\src\ecc.cpp:540: error: (-7) The algorithm stopped before its convergence. The correlation is going to be minimized. Images may be uncorrelated or non-overlapped in function cv::findTransformECC
"""
The above exception was the direct cause of the following exception:
error Traceback (most recent call last)
<ipython-input-2-8126994b559e> in <module>()
321 print("Alinging images. Depending on settings this can take from a few seconds to many minutes")
322 # Increase max_iterations to 1000+ for better results, but much longer runtimes
--> 323 warp_matrices, alignment_pairs = imageutils.align_capture(capture, max_iterations=1000)
324 
325 print("Finished Aligning, warp matrices:")
c:\micasense\imageprocessing-master\micasense\imageutils.py in align_capture(capture, ref_index, warp_mode, max_iterations, epsilon_threshold)
115 #multiprocessing.set_start_method('spawn')
116 pool = multiprocessing.Pool(processes=multiprocessing.cpu_count())
--> 117 for i,mat in enumerate(pool.imap_unordered(align, alignment_pairs)):
118 warp_matrices[mat['match_index']] = mat['warp_matrix']
119 print("Finished aligning band {}".format(mat['match_index']))
C:\Anaconda3\envs\micasense_pnnl\lib\multiprocessing\pool.py in next(self, timeout)
733 if success:
734 return value
--> 735 raise value
736 
737 __next__ = next # XXX
error: OpenCV(3.4.1) C:\bld\opencv_1520732670222\work\opencv-3.4.1\modules\video\src\ecc.cpp:540: error: (-7) The algorithm stopped before its convergence. The correlation is going to be minimized. Images may be uncorrelated or non-overlapped in function cv::findTransformECC"
When i try to align the photos of the 5 bands, some o the sample images throw this error. Is it possible to be same problem with the sample data. Or a python error?
My code can do align over the iteration but after 8 or 9 files i get this error. When i test my code with a single image i get the same error with some files. 
I already replaced all the sample data from the zip file, but still get the same error.
Thank you.

poynting, 2018-07-13:
Hi @filmfer:  

This is not the recommended approach.  We've described the pros and cons of our alignment method as well as the cases it is and isn't expected to work in detail in the Alignment Tutorial at https://micasense.github.io/imageprocessing/Alignment.html

This matching code is simply for demonstration and as described in the link above isn't expected to be robust all images in a set; however, as the text describes, it is only necessary to find a good alignment for one image in a flight. Once a good alignment matrix is found, simply use that alignment for all images in the flight. 

Pick an image with good edge features in all bands (such as the demonstration image has), let the algorithm run for 1000 or more iterations, and verify the registration using the provided notebook. Then run the alignment part of the code across all of the images.  Try a few different images, I'm sure you'll find one that finds matches across bands. If not, take a new image over a building or a parking lot where there are lots of edges that appear in all bands, then use that to find your alignment transform.

closed 2018-07-13 by poynting

poynting, 2019-03-23:
@csomerlot please see the new altum-support branch, specifically the Alignment and Batch Processing notebooks. They walk through this process and simplify it substantially from before.

```
git fetch
git checkout altum-support
```