micasense/imageprocessing #13: About error on camera firmware.

xurong1981, 2018-07-20:
When I use micasense library to deal with the data captured by our RedEdge-M, but the following error occurred.
----
ValueError: Library requires images taken with camera firmware v2.1.0 or later. Upgrade your camera firmware to use this library.
----

Is there any method to solve this problem by using old version of the source ?
Or the only method is to update the camera firmware ?

poynting, 2018-07-21:
There is no intention of having the library support older firmware, as
older firmware doesn't store all of the elements of the full radiometric
model. That version is quite old already, we recommend strongly that all
users update the firmware to the most recent version.

Justin

xurong1981, 2018-07-23:
I see. Thank you so much for your reply.

closed 2018-07-25 by poynting