micasense/imageprocessing #14: Panel snippet identifies wrong area in image

smescarzaga, 2018-07-25:
I've finished running through the environment setup for the Micasense utilities. After successfully running the test calibration panel code that identifies the panel, I decided to try in on an image I've taken with the RedEdge-M. The code successfully runs and it correctly identifies the QR code on the panel, however it draws the blue box around the wrong portion of the panel (in the grass or elsewhere) as opposed to the calibrated surface it self. I've replicated this error on a number of different panel images from different dates. Below is an output of that code:

[image]

The only thing I've changed from the example panels code snippet are the imagePath and imageName variables:

```
imagePath = 'D:/Alaska_UAV_Flights_&_Specs/BRW_CALM/20180721/MicaSense/0025SET/000'
imageName = glob.glob(os.path.join(imagePath,'IMG_0004_4.tif'))[0]

```

poynting, 2018-07-25:
It looks like the code orientation changed in RP04 panels, but the size also changed. This isn't accounted for by the library currently.  I'll be out for a couple of weeks but the below might get you through until I can update the library to support the newer panels.

There's a commented line in the panel.py `panel_corners` method (see below).  Try un-commenting it and setting rotation to 2. This might get you the right orientation. However you'll also need a smaller panel area for the newer panel.  Try reducing the size of the square defined by `reference_panel_pts`.

```
# a rotation may be necessary for *very* old first gen panels
#rotation = 1; reference_qr_pts = np.roll(measuredQrPts, rotation, axis=0) 
```

smescarzaga, 2018-07-25:
Thanks much for your speedy response!

Un-commenting the `rotation` variable results in another error:

``````
NameError                                 Traceback (most recent call last)
<ipython-input-4-f2ec7f8b41ed> in <module>()
     11     raise IOError("Panel Not Detected! Check your installation of pyzbar")
     12 else:
---> 13     panel.plot(figsize=(8,8));
     14 
     15 print('Success! Now you are ready for Part 1 of the tutorial.')

~\Documents\GitHub\imageprocessing\micasense\panel.py in plot(self, figsize)
    175 
    176     def plot(self, figsize=(14,14)):
--> 177         display_img = self.plot_image()
    178         fig, ax = plt.subplots(figsize=figsize)
    179         ax.imshow(display_img)

~\Documents\GitHub\imageprocessing\micasense\panel.py in plot_image(self)
    165         if self.panel_detected():
    166             cv2.drawContours(display_img,[self.qr_corners()], 0, (255, 0, 0), 3)
--> 167         cv2.drawContours(display_img,[self.panel_corners()], 0, (0,0, 255), 3)
    168 
    169         font = cv2.FONT_HERSHEY_DUPLEX

~\Documents\GitHub\imageprocessing\micasense\panel.py in panel_corners(self)
    100                                       dtype=np.int32)
    101         # a rotation may be necessary for *very* old first gen panels
--> 102         rotation = 2; reference_qr_pts = np.roll(measuredQrPts, rotation, axis=0)
    103 
    104         src = np.asarray([tuple(row) for row in reference_qr_pts[:3]], np.float32)

NameError: name 'measuredQrPts' is not defined
``````

Sorry if this issue is a basic Python (still learning), but I'm not sure where to define/introduce `measuredQrPts`

maykef, 2018-07-25:
Hey, 

I had the same problem. You do need to un-comment the line mentioned by Justin. Change 'measuredQrPoints' by 'reference_qr_pts' and the rotation number between 0-4 until the blue square falls into the panel area.

maykef, 2018-07-25:
[Panels.pdf](https://github.com/micasense/imageprocessing/files/2227076/Panels.pdf)

maykef, 2018-07-25:
Also, as Justin mentioned, you need to change the area of your blue square so it falls into your much smaller reference panel.

smescarzaga, 2018-07-25:
Hey guys,

Thank you very much for your suggestions. I tweaked the rotation and box size and it looks pretty good.

[image]

yellowpuppy, 2018-08-11:
I have solved this problem by redefining the affine coordinate

maykef, 2018-08-11:
Mind to elaborate a little bit further?

smescarzaga, 2018-08-15:
Strangely enough, it looks like Agisoft Photoscan has already incorporated the new reference panels into their calibration workflow for Pro 1.4.

poynting, 2018-08-19:
I haven't been able to get back to this, but it looks like everyone here has found solutions. If someone has come up with a general solution please create a pull request.  I think a good general solution would be to use a smaller area as defined above, and to try all 4 roll operations of the panel coordinates. The statistics of the resulting area should make it clear which area is the panel: it should have a quite low standard deviation relative to non-panel areas.

Another approach that may be more stable is simply to look at the revision of the panel: RP04 panels should have the 'new' transform while older panels will have the older transform.

Also as some of you may have noticed, for newer (RP04) panels the reflectance calibration is built into the serial number.  The tail end of the panel above "_D400508100518" provides that the panel calibration can be reasonably approximated by a line through 400nm, 50.8% reflectance and 1000nm, 51.8%.  This line can then be used to automatically determine the panel reflectance for each band.

andbrs, 2019-03-07:
Hi, I have the same problem but I have tried the two solutions proposed above but none have worked. This has to do with why I use python 3.7? or you can explain in more detail. Thank you

poynting, 2019-03-19:
Hi All, I believe this should now be fixed in the altum-support branch. It'll be a little while before that branch is well tested and merged into master, but in the meantime you should be able to use it to fix these panel mis-identification issues.  Please let me know if you still see issues using that branch.

closed 2019-04-09 by poynting