micasense/imageprocessing #16: merged images generate ghosting

SmithPeng, 2018-08-17:
I used micasense images for mosaic through [opendronemap](https://github.com/OpenDroneMap/OpenDroneMap) , and opendronemap only received three-band '*.jpg' images. In order to get RGB mosaic images and NDVI mosaic images I had done operations, which is as follows:

1. merged blue/green/red images into one
2. calculated reflectance for red/nir/rededge band , and merged them ,then plus 255.0 for each pixel to show images in *.jpg
3. stitching in opendronemap

but ghosts appeared in merged images, and I stop in step 2 to find how to solve the problem. merge results are as follows.
step 1 result:
[image]
step 2 result:
[image]

anybody know how to solve the problem?

In addition, specific panel calibration aim at this specific images or specific aircraft or specific scene or others?
`    # Our panel calibration by band (from MicaSense for our specific panel)
    panelCalibration = { 
        "Blue": 0.67, 
        "Green": 0.69, 
        "Red": 0.68, 
        "Red edge": 0.67, 
        "NIR": 0.61 
    }`

poynting, 2018-08-19:
Hi @SmithPeng.  Without more information about what you're doing to run the code or what parameters you're using it will be quite difficult to help.  In your steps, it's unclear why step 1 is necessary; radiometric calibration and alignment is the purpose of step 2, it should be unnecessary to combine the images before step 2.

Please read over the alignment tutorial, specifically the section here https://micasense.github.io/imageprocessing/Alignment.html#Unwarp-and-Align

Also, the same tutorial has an example of how to export the images to jpg:
https://micasense.github.io/imageprocessing/MicaSense%20Image%20Processing%20Tutorial%201.html

(however if you want to calculate indices like NDVI from the JPEG, you will want to remove the gamma correction, as it will ruin the results.

Also, for a general overview of radiometry and reflectance calibration, see here https://micasense.github.io/imageprocessing/MicaSense%20Image%20Processing%20Tutorial%201.html

From those resources you'll find two important things:  the panel calibration is specific to the panel and should be performed for every flight, and the alignment function provided has inputs which can be modified to run longer and exert more effort to find a good transform.

The full list of tutorials is on the main page of the repository, please read them over and open a more specific ticket if there's a specific issue with the repository code or it's usage we can help with.

closed 2018-08-19 by poynting

SmithPeng, 2018-08-21:
I followed the operation showed in Alignment.html,but error occurred.

> warp_matrices, alignment_pairs = imageutils.align_capture(capture, max_iterations=10)
  File "./micasense/imageutils.py", line 101, in align_capture
    capture.images[ref_index].reflectance()).astype('float32')
  File "/home/pengyaoyao/project/uav_registration/imageprocessing/micasense/image.py", line 135, in reflectance
    raise RuntimeError("Provide a band-specific spectral irradiance to compute reflectance")
RuntimeError: Provide a band-specific spectral irradiance to compute reflectance