micasense/imageprocessing #17: irradiance not defined

SmithPeng, 2018-08-27:
I followed the operation showed in Alignment.html,but error occurred.

> warp_matrices, alignment_pairs = imageutils.align_capture(capture, max_iterations=10)
File "./micasense/imageutils.py", line 101, in align_capture
capture.images[ref_index].reflectance()).astype('float32')
File "/home/pengyaoyao/project/uav_registration/imageprocessing/micasense/image.py", line 135, in reflectance
raise RuntimeError("Provide a band-specific spectral irradiance to compute reflectance")
RuntimeError: Provide a band-specific spectral irradiance to compute reflectance

irradiance is defined by what? I haven't found.

poynting, 2018-08-27:
It's hard to know what's going on without seeing the failing code.  Is it possible that you haven't implemented the radiometric corrections that are implemented in cell #2 of the alignment tutorial prior to running `align_capture`?  

If you have questions about the radiometric corrections, please refer to the earlier tutorials where they are described in detail.  It would be a good start to ensure you can run the Alignment notebook on the provided data. Then substitute your own data, and try to determine what might be different.

https://github.com/micasense/imageprocessing/blob/master/Alignment.ipynb

In the future an issue report should always contains at minimum the following elements:

1. Title/Description
2. Environment (including all code that is failing)
3. Steps to reproduce
4. Expected Result(s)
5. Actual Result(s)

SmithPeng, 2018-08-28:
I have  run the Alignment notebook on the provided data

poynting, 2018-08-28:
It may not be obvious from the alignment tutorial, but the library lazy-computes reflectance images within a `Capture` object only when they are needed.  In the tutorial, the 

```python
capture.plot_undistorted_reflectance(panel_irradiance)
``` 
method ensures this computation is done. If plotting is not necessary, instead call 
```python
capture.compute_reflectance(panel_irradiance)
```
where `panel_irradiance` is computed using panel image radiance and the provided panel reflectance values (which are different for every panel) by calling
```
capture.panel_irradiance(panel_reflectance)
``` 
and `panel_reflectance` is a list of panel reflectance values in the range 0-1 for each band, in RedEdge band_index order (not wavelength order).

SmithPeng, 2018-08-28:
Firstly, the problem is how to calibrate raw images from the camera into reflectance maps

(1).https://github.com/micasense/imageprocessing/blob/master/MicaSense%20Image%20Processing%20Tutorial%201.ipynb

As (1) described, from raw images to radiance worked as following:
`import micasense.utils as msutils
radianceImage, L, V, R = msutils.raw_image_to_radiance(meta, imageRaw)`
convert radiance to reflectance worked as following:
`# Select panel region from radiance image , some codes are omitted
meanRadiance = panelRegion.mean()
panelReflectance = panelCalibration[bandName]
radianceToReflectance = panelReflectance / meanRadiance
reflectanceImage = radianceImage * radianceToReflectance`

(2).https://github.com/micasense/imageprocessing/blob/master/Alignment.ipynb

As (2) described,from raw images to reflectance worked as following:
`panel_irradiance = panelCap.panel_irradiance(panel_reflectance_by_band)
capture.plot_undistorted_reflectance(panel_irradiance)`

from raw images to radiance, in 'panel_irradiance()',
`irradiance_list = []
 for i,p in enumerate(self.panels):
        mean_irr = p.irradiance_mean(reflectances[i])
        irradiance_list.append(mean_irr)`
where
`def irradiance_mean(self, reflectance):
        radiance_mean, _, _, _ = self.radiance()
        return radiance_mean * math.pi / reflectance`
convert radiance to reflectance worked as following,in 'plot_undistorted_reflectance',
`self.reflectance(irradiance_list)`

and in (2), 'self.radiance()' equal to 'msutils.raw_image_to_radiance()' in (1), reflectances[i] in (2)  equal to 'panelCalibration[bandName]' in (1), self.panels in (2) equal to panelRegion in (1) . 
In (1), transform coefficient from radiance to reflectance  is radianceToReflectance , in (2), transform coefficient is irradiance_list, is  'mean_irr' in (2) equal to 'meanRadiance' in (1)? I haven't seen something like 'panelRegion.mean()', how could 'radiance_mean' in (2) represent the mean of radiance image?

SmithPeng, 2018-08-28:
Secondly, you are right ! 
**capture.plot_undistorted_reflectance(panel_irradiance)**

poynting, 2018-08-28:
I think your first question is answered in https://github.com/micasense/imageprocessing/blob/master/micasense/panel.py#L144
where `self` here is a `Panel` object, so it has a defined region if a panel is found in the image or if the corners are provided at initialization.
```
    def radiance(self):
        radiance_img = self.image.undistorted(self.image.radiance())
        return self.region_stats(radiance_img,self.panel_corners())
```

Your second issue is likely that you're re-running the alignment code in the same notebook without re-starting the kernel.  `multiprocessing.set_start_method('spawn')` can only be called once in a program.  

The solution is probably to "restart and run all" in your notebook.  This is a terrible feature of jupyter notebooks; they save state and tend to hide it from the developer.  Many times notebook problems can be fixed with "restart and run all."

SmithPeng, 2018-08-29:
first question is OK

second issue is solved, too. 

When using **spawn** I should guard the part that launches the job in **if __name__ == '__main__':****.set_start_method** should also go there, and everything  run fine.

closed 2018-08-29 by poynting