micasense/imageprocessing #19: A problem about calculating DLS_irradiance,  a bug?

yellowpuppy, 2018-09-08:
As the introduction in tutorial3.ipynb ,  in Cell 3, 
`    # compute irradiance on the ground using the solar altitude angle
    dls_irr = untilted_direct_irr * (percent_diffuse + np.sin(solar_elevation))`
The DLS_irr was calculated by "sin" of solar_elevation.
However, in capture.py  , line 162
`            # compute irradiance on the ground using the solar altitude angle
            ground_irr = untiltied_direct_irr * (percent_diffuse + np.cos (self.solar_elevation))`
The same var was calculated by "cos" of solar_elevation.

This is a bug and which one is correct?  Anyone can help me?

poynting, 2018-09-08:
> The results came from running the code in tutorial 3 and
> “Capture.dls_irradiance()” function in capture.py are totally different.

Good catch. The library code was updated to fix that bug but we missed it
in the tutorial. The library code should be the correct implementation.

yellowpuppy, 2018-09-09:
Thank you @poynting ,I saw an issue that had been shut down (#11 ).
It is seem like we should use sin(solar_elevation) to correct the direct irradiance
Therefore, the tutorial code should be the correct but the library code maybe wrong.

ComteDeLooz, 2018-09-19:
Hi,
The solar elevation is the altitude of the sun above the horizon (= 90° - sun zenith angle). Therefore, I think we should indeed use **np.sin(solar_elevation)**. If not, this means if the sun is just at the horizon (elevation of 0°), it contributes more to the irradiance on the ground (np.cos(0°) = 1) than if the sun is at zenith (np.cos(90°) = 0) which does not seem logic to me. But maybe I am missing something...

poynting, 2018-09-25:
Fixed in c1483c474f24b408a4028ad99c59a812756a001b

closed 2018-09-25 by poynting