micasense/imageprocessing #20: Image Processing Setup

mattwake, 2018-09-24:
I am currently trying to run the python script for testing to see if all of the modules import on a windows machine. I am using Spyder Python 3.6 to run the script.   I am having trouble with the exiftool. I downloaded the exiftool.exe file and stored it in my anaconda3\lib\site-packages\exiftool\exiftool.exe and added a system path variable to that location. I have also tried storing it in this location as well C:\exiftool\exiftool.exe and added a system path to there.  I was able to set up the MicaSense Environment in Anaconda Prompt and run the yml file. I also had to pip install modules pyzbar and mapboxgl. When I run the Setup Script I get the following output:

**"Successfully imported all required libraries.

Exiftool isn't working. Double check that you've followed the instructions above.
The execption text below may help to find the source of the problem:

module 'exiftool' has no attribute 'ExifTool'"**

Any idea on how to fix this?? I have been searching all over for a fix.

poynting, 2018-09-25:
If you had to pip install both `pyzbar` and `mapboxgl` that would imply that none of the pip dependencies in the yml installed properly. You'll probably need to activate your conda environment and manually install the last pip dependency as well, which is the pyexiftool wrapper:
```
pip install git+https://github.com/smarnach/pyexiftool.git#egg=pyexiftoolpy
```
We should add one more import to the setup notebook which I think would catch this
```
import exiftool
```

mattwake, 2018-09-25:
Thank you very much! I figured out it was the environment issue since Spyder was installed under the base environment. I installed it under the Micasense environment and the setup file executed properly.

closed 2018-09-25 by mattwake