micasense/imageprocessing #22: Colorbar covers Image

maltepaul, 2018-10-24:
I have problems with the plotting tools like plotwithcolorbar. The colorbar always covers the image.
[image]
Not really an expert in Python so I have no idea what the problem might be.
Anyone knows how to fix this?

poynting, 2019-03-15:
I can't recreate this, can you provide more information?

poynting, 2019-06-06:
I'm going to close this due to lack of information and inability to reproduce on any platform. I suspect this issue has to do with incompatible versions of some of the underlying libraries, but that's just a guess.  Try using the conda installation process to create a new environment and load compatible libraries.

closed 2019-06-06 by poynting