micasense/imageprocessing #24: pytest.py failed test

Mercuron, 2018-11-29:
Hello!
Can you please solve this problem?
My command `pytest.py .` failed, all packages are installed
`~/imageprocessing$ conda activate micasense
(micasense)~/imageprocessing$ pytest .
============================= test session starts ==============================
platform linux -- Python 3.6.7, pytest-4.0.1, py-1.7.0, pluggy-0.8.0
rootdir: /home/antigr/imageprocessing, inifile: pytest.ini
collected 79 items                                                             

tests/test_capture.py FFFFFFFFFFFFFFFF                                   [ 20%]
tests/test_dls.py F.......                                               [ 30%]
tests/test_image.py ...........                                          [ 44%]
tests/test_imageset.py FFF                                               [ 48%]
tests/test_metadata.py ...............................                   [ 87%]
tests/test_panel.py FFFFFFFFFF                                           [100%]

=================================== FAILURES ===================================
_______________________________ test_from_images _______________________________

    def test_from_images():
>       imgs = [image.Image(fle) for fle in file_list()]
E       _pytest.warning_types.RemovedInPytest4Warning: Fixture "file_list" called directly. Fixtures are not meant to be called directly, are created automatically when test functions request them as parameters. See https://docs.pytest.org/en/latest/fixture.html for more information.

tests/test_capture.py:47: RemovedInPytest4Warning
`

... and there was more and more same issues
i tried to reinstall all but it didnt help,
thank!

poynting, 2018-11-29:
Looks like an issue related to depreciation in Pytest 4.0.1 will require some code updates.  In the meantime try to use pytest 3.6.0 instead and see if that works.

`conda install pytest=3.6.0`

Mercuron, 2018-11-29:
Thanks for a fast answer!
at ubuntu 1804 miniconda 4.x had changed activate function, i had some problems with that.
So you can not just do `conda activate micasense`
I did not right solve, but this commands can win it 
```
source activate micasene
source deactivate
conda activate micasense
```
I dont know is it bug or error, but its works and may be helpful

poynting, 2019-03-21:
For now I've just solved this by pinning the pytest version in the altum-support branch. I'll close this when that is merged.

poynting, 2019-03-28:
Changed my mind on this due to some other conda environment issues forcing me to update to the latest version of pytest, I went ahead and fixed it.  Will be in the altum-support branch.

poynting, 2019-04-09:
Fixed in altum-support. Will be merged to master soon.

closed 2019-04-09 by poynting