micasense/imageprocessing #25: why we cant align pic with math?

Mercuron, 2018-11-30:
Hello!
why we cant calculate the exact pixels count to cut images and just using numpy to join them? There was text about temperature and other factors, but is it really affect it so much?
I think we got all values, isnt it? i tried to use 100, 1000, 10000 iterations and got the same result. How can i improve alignment quality ? 
[image]
[image]
[image]
[image]

poynting, 2019-03-19:
Hi @Mercuron - you're in luck, we have image alignment using lots of math now in the altum-support branch.  Check out the Alignment-RigRelatives,ipynb notebook and point it to your own images.  Note, the images need to be from at least firmware version 3.4.0 or they need to have the RigRelatives tags added as described in https://support.micasense.com/hc/en-us/articles/360006368574-Modifying-older-collections-for-Pix4Dfields-support

poynting, 2019-04-09:
Fixed in the altum-support branch via the Alignment-RigRelatives tutorial.  Closing this in preparation to merge that branch to master soon.

closed 2019-04-09 by poynting