micasense/imageprocessing #28: OSError: [WinError 10038] An operation was attempted on something that is not a socket

AJFeng, 2019-01-23:
When I run the Testing Installation,
Successfully imported all required libraries.
Successfully executed exiftool.

But when I run the Part 1 of the tutorial,

OSError                                   Traceback (most recent call last)
<ipython-input-4-b10e581ba09b> in <module>()
      4     exiftoolPath = 'C:/exiftool/exiftool.exe'
      5 # get image metadata
----> 6 meta = metadata.Metadata(imageName, exiftoolPath=exiftoolPath)
      7 cameraMake = meta.get_item('EXIF:Make')
      8 cameraModel = meta.get_item('EXIF:Model')

~\Downloads\imageprocessing-master\imageprocessing-master\micasense\metadata.py in __init__(self, filename, exiftoolPath)
     42             raise IOError("Input path is not a file")
     43         with exiftool.ExifTool(self.exiftoolPath) as exift:
---> 44             self.exif = exift.get_metadata(filename)
     45 
     46     def get_all(self):

C:\ProgramData\Anaconda3\lib\site-packages\exiftool.py in get_metadata(self, filename)
    270         documentation of :py:meth:`execute_json()`.
    271         """
--> 272         return self.execute_json(filename)[0]
    273 
    274     def get_tags_batch(self, tags, filenames):

C:\ProgramData\Anaconda3\lib\site-packages\exiftool.py in execute_json(self, *params)
    254         """
    255         params = map(fsencode, params)
--> 256         return json.loads(self.execute(b"-j", *params).decode("utf-8"))
    257 
    258     def get_metadata_batch(self, filenames):

C:\ProgramData\Anaconda3\lib\site-packages\exiftool.py in execute(self, *params)
    225         fd = self._process.stdout.fileno()
    226         while not output[-32:].strip().endswith(sentinel):
--> 227             inputready,outputready,exceptready = select.select([fd],[],[])
    228             for i in inputready:
    229                 if i == fd:

OSError: [WinError 10038] An operation was attempted on something that is not a socket.

Could anyone help me with this error? Thanks.

ComteDeLooz, 2019-03-05:
I had the same error message "operation on something that is not a socket [WinError 10038]". With me, it was linked to the ExifTool version I downloaded from the website of Phil Harvey. This version was 11.29. When I installed an earlier version of ExifTool (11.2.0.0) together with the pyexiftool tarball (smarnach-pyexiftool-v0.2.0-0-g3db3764.tar.gz), things worked out.

poynting, 2019-03-15:
Based on the above I'm going to close this, as it appears to be an exiftool/pyexiftool issue, not a library code problem.  If you find that to not be the case and have a fix that we can implement here, let me know.

closed 2019-03-15 by poynting

AJFeng, 2019-06-14:
https://github.com/smarnach/pyexiftool/issues/26

This one works for me.