micasense/imageprocessing #29: Correction of Altum identical to RedEdge?

swawa, 2019-02-14:
Hi,
a question: I am working with Micasense Altum and wonder if the here offered image processing snippets also work for this camera type?
Also in terms of distortion and vignette correction, and the radiometric correction?

Do I have to take care of any differences between the two camera types?
Thanks!
balumbala

poynting, 2019-02-26:
It's not exactly the same due to the addition of the thermal sensor, but for the narrowband data only it is very similar.  We're working on some code updates to support Altum transparently through this code base. I don't have an ETA but we have the code working locally and we're now working through the test and verification changes.

swawa, 2019-02-26:
Ok, thanks! I am really curious about it. Most of the other snippes work in principle. There're really good commented and the programming style is good to follow. Seldom seen such a good structure :-)
The only thing that need adaption is if people don't work with Jupyter but with Spider or other Python surfaces. But that is adoptable.
Good work!

poynting, 2019-03-19:
Hi @swawa - I've committed the altum-support branch. I haven't generated the documentation (html files) yet, but it should support Altum in the same way as RedEdge.  Probably the easiest test is to run the Alignment-RigRelatives.ipynb and change the paths to point to two of your own files - a panel image and an aerial image, and see if things work for you.

swawa, 2019-03-19:
Hi Justin,

wow, good news! I will download the branch and test it.
It may take a week as I do have to finish some other work before.

Thanks for that fast support.

MOnika

swawa, 2019-03-19:
Hi Justin again,

sorry is the branch on github micasense? There I can’t find it

best wishes,
monika

poynting, 2019-03-20:
Yes: https://github.com/micasense/imageprocessing/tree/altum-support?files=1

Or try  

git fetch
git checkout altum-support

poynting, 2019-04-09:
Closing this in preparation for merging the altum-support branch to master in the next couple of days.

closed 2019-04-09 by poynting