micasense/imageprocessing #30: the "%" symbol usage in these scripts

swawa, 2019-02-21:
Hi,
in some python scripts appear “%” as a first symbol in line. (my) Python can’t read that. I am familiar with the “%” symbol in order to apply a modulus operation that is placed between to variables/values.
What does it mean here? do I need a special package to use it?

Like on https://micasense.github.io/imageprocessing /MicaSense%20Image%20Processing%20Tutorial%201.html 
paragraph
”Load a directory of images into an ImageSet”

Or on 
https://micasense.github.io/imageprocessing/Images.html
paragraph
“images”

Best wishes

swawa, 2019-02-21:
ok, in the mean time I've learned that the "%" sign is also used as quick reference in Jupyter notebooks.
My question ist now: I would like to adopt the features to a Python 3.6 script.

what is missing for an adequate replacement
instead of calling
images_dir = os.path.expanduser(os.path.join('~','Downloads','RedEdgeImageSet','0000SET'))
%time imgset = imageset.ImageSet.from_directory(images_dir, progress_callback=update_f)
from a Jupyter notebook

I will call this in a script (still erroneous):
import time
start = time.time()
my_list = [i for i in range(1000000)]
images_dir = os.path.expanduser(os.path.join('~','Downloads','RedEdgeImageSet','0000SET'))
imageset = imageset.ImageSet.from_directory(images_dir)
elapsed_time=(time.time()-start)
print(elapsed_time)

Best wishes

poynting, 2019-02-26:
The timing is just there to show how long that portion of the script runs, it's not required. If you're running the code outside of a Jupyter notebook you can just delete that line.

closed 2019-02-26 by poynting

swawa, 2019-02-28:
Ok, thanks :-)