micasense/imageprocessing #31: exiftool not working

avakanski, 2019-02-28:
I tried to install micasense on Windows and I followed all instructions at least 10 times, but I can not get exiftool to work. 
I got the message all the time: "No module named 'exiftool'"

I installed exiftool.exe into a folder C:\exiftool\exiftool.exe, and I set the path to this folder. 

The exception text in the is: 'exiftool' is not defined"

poynting, 2019-02-28:
I think "exiftool is not defined" would generally mean that pyexiftool (the exiftool python wrapper) didn't install properly, or the environment isn't activated.  It is a bit odd that it doesn't fail on import though.

```
(base) C:\Users\Justin>activate micasense
(micasense) C:\Users\Justin>python -c "import exiftool; print(exiftool.ExifTool())"
<exiftool.ExifTool object at 0x0000018BDA15A1D0>
```

If your issues persist you might get better support from https://github.com/smarnach/pyexiftool/issues or someone else lurking here as I don't have a ton of python packaging experience.

avakanski, 2019-02-28:
Hi Justin,
Thanks a lot for the reply. I did activate the micasense environment, so it
is likely that pyexiftool didn't install properly. I will check that, it
makes a lot of sense.
If it doesn't work I will contact the support.
Much appreciated,
Alex

avakanski, 2019-02-28:
It seems that one of the problems may be the way the folders are defined in Windows and Linux.
 The path to exiftool was read as "C:\\exiftool\\exiftool.exe" and the system couldn't find it, but when I manually changed it to "C:\exiftool\exiftool.exe" it worked properly. 
Now I have get other error messages of the type "operation on something that is not a socket" [WinError 10038].
It may also be related to the Windows OS that I am using.

ComteDeLooz, 2019-03-05:
I had the same error message "operation on something that is not a socket [WinError 10038]". With me, it was linked to the ExifTool version I downloaded from the website of Phil Harvey. This version was 11.29. When I installed an earlier version of ExifTool (11.2.0.0) together with the pyexiftool tarball (smarnach-pyexiftool-v0.2.0-0-g3db3764.tar.gz), things worked out.

avakanski, 2019-03-05:
Thank you so much for your reply. I will give a try and hopefully it will work for me. I spent 2 days on trying to make the exiftool works and I almost gave up. Much appreciated, Alex

From: ComteDeLooz 
Sent: Tuesday, March 05, 2019 8:35 AM
To: micasense/imageprocessing 
Cc: avakanski ; Author 
Subject: Re: [micasense/imageprocessing] exiftool not working (#31)

I had the same error message "operation on something that is not a socket [WinError 10038]". With me, it was linked to the ExifTool version I downloaded from the website of Phil Harvey. This version was 11.29. When I installed an earlier version of ExifTool (11.2.0.0) together with the pyexiftool tarball (smarnach-pyexiftool-v0.2.0-0-g3db3764.tar.gz), things worked out.

andbrs, 2019-03-05:
hello I had the same problem, the solution is to install through command window the exiftool with the following lines
pip install mapboxgl
pip install git+https://github.com/smarnach/pyexiftool.git#egg=pyexiftoolpy
in the anaconda prompt

avakanski, 2019-03-06:
Thanks, I tried your suggestion and I received a message that the requirements are already satisfied. Thank you in any case

From: andbrs 
Sent: Tuesday, March 5, 2019 11:35 AM
To: micasense/imageprocessing 
Cc: avakanski ; Author 
Subject: Re: [micasense/imageprocessing] exiftool not working (#31)

hello I had the same problem, the solution is to install through command window the exiftool with the following lines
pip install mapboxgl
pip install git+https://github.com/smarnach/pyexiftool.git#egg=pyexiftoolpy
in the anaconda prompt

poynting, 2019-03-15:
Based on the comments above, I don't think this is a library issue as much as an issue with newer versions of exiftool. You might check the exiftool forums or the pyexiftool github.  I'm going to close this as I don't know of any library changes that could fix exiftool.

closed 2019-03-15 by poynting