micasense/imageprocessing #32: Correction of lens distortion not working

andbrs, 2019-03-05:
greetings to all, I am trying to work with the micasense libraries but I have a problem with tutorial 1 in the part of correct for lens distortions, when processing the image the result is a blurry photo. I have a micasense rededge m with software 4.2.2 and if I change photo I get an error very similar to the error of this post https://github.com/micasense/imageprocessing/issues/3
The strange thing is that if I use the example photos provided by micasense (data folder) with version 2.1 the program works perfectly. I appreciate any help and also attach the photo with the problem after of correct for lens distortions.

[image]

poynting, 2019-03-05:
Hi @andbrs, thanks for creating a new issue.  It looks like the problem is that the changes made by in #3 were inadvertently reverted in a later update.  Can you verify that if you replace your utils.py with this version it fixes your issues?

https://github.com/micasense/imageprocessing/blob/11b24ff77c4ed57c6137526d84b255cbf1faa13b/micasense/utils.py

andbrs, 2019-03-05:
It works! Thank you, replacing the file works correctly with all version of fimware. Thanks again

poynting, 2019-03-05:
Cool, I'm working on some other stuff at the moment, but I'll update this when I have a chance. In general, the library-like code should be more stable as we've wrapped up most of the functionality there, but certainly the tutorial should work properly.

andbrs, 2019-03-06:
Great!!!!, thanks for the help.

poynting, 2019-03-19:
Hi @andbrs - I believe I have fixed this in the altum-support branch. Can you check out that branch and let me know if it resolves this issue for you?

closed 2019-04-09 by poynting