micasense/imageprocessing #34: mala identificación del panel de calibracion

andbrs, 2019-03-08:
hello, I'm testing the code identification code qr calibration panel, this process does well because it identifies the serial number, the problem is that it marks the blue box in a different area. here I upload two captures with different images presenting the error. Thank you very much for the help 

![dawwd](https://user-images.githubusercontent.com/9252592/54031593-30fb3e00
[image]
-417d-11e9-8a77-4f480f5b8308.png)

poynting, 2019-03-15:
Hi @andbrs, can you attach or link to the actual files that are failing. The location and orientation of the blue square is very suspicious that something isn't right.

andbrs, 2019-03-15:
Hello, thank you for answering the truth is that you fail with all the images I've taken. The only solution has been to modify the library but this solution is not very convenient because if the panel is a little inclined the picture is placed in a place that does not correspond.

link photos
https://drive.google.com/file/d/1sjVs87Obi5aElgPQHKolGdVtRh0LGybF/view?usp=sharing
https://drive.google.com/file/d/1S7o7RiOjFZJgfSeM8WrI9ls6RHOe_SRi/view?usp=sharing
https://drive.google.com/file/d/1aa40rTYNNAt5-2dT3S6HZy3BxK_dEOrf/view?usp=sharing
https://drive.google.com/file/d/1pMsFABOZ4saiOsqGY9UpR83imcCk4uWC/view?usp=sharing
https://drive.google.com/file/d/1dX6_aPQRW-LtHtxCwcEDZ1m6AlvPqglp/view?usp=sharing
https://drive.google.com/file/d/1VsoI82Z2V2KWEEC6EN9xPzXv3DriABLM/view?usp=sharing

Translated with www.DeepL.com/Translator

poynting, 2019-03-19:
Hi @andbrs - I believe this is fixed in the altum-support branch - can you checkout that branch and let me know if it solves your issues?

andbrs, 2019-03-26:
I'm sorry I'm late, but I've been a little busy with other things. I'm happy to corroborate the solution, but could you tell me what the altum support branch is? My camera is an edge network

poynting, 2019-03-26:
in your code checkout directory,

git fetch
git checkout altum-support

poynting, 2019-03-26:
Once you have the new branch, a simple way to test this is to run the Panels.ipynb notebook and point it at your files. When I did that, everything worked on each of them.

You can also try to run the panels tests on your system to verify the panel finding works on the provided files.
```
pytest .
```
to run all tests, which can take a few minutes, or
```
pytest tests/test_panel.py
```
to run just the panels unit tests.  Do please make sure you're running pytest version 3.6.0 as 4.x is known to not work currently.  If using conda for package management you can do this with
```
conda install -c conda-forge pytest=3.6.0
```

andbrs, 2019-04-01:
> in your code checkout directory, git fetch git checkout altum-support
> […](#)

I'm sorry if I'm a little slow about this, but I've been very busy, but I can already look up what you tell me. I've been looking for those addresses in my code and I can't find it (in the code there's no reference to the altum or at least the one I clone). Could you be a little more specific how do I check if it's already fixed? Thank you for your patience

poynting, 2019-04-01:
Open the checkout directory of the code in your command line. Type

git fetch
git checkout altum-support

andbrs, 2019-04-01:
shows me this error

error: Your local changes to the following files would be overwritten by checkou         t:                                                                                               Alignment.ipynb                                                                          MicaSense Image Processing Setup.ipynb                                                   MicaSense Image Processing Tutorial 1.ipynb                                              MicaSense Image Processing Tutorial 2.ipynb                                              MicaSense Image Processing Tutorial 3.ipynb                                              Panels.ipynb                                                                             micasense/metadata.py                                                                    micasense/panel.py                                                                       micasense/utils.py                                                               Please commit your changes or stash them before you switch branches.                     Aborting
--

poynting, 2019-04-01:
I'm sorry, I very much want to help but I don't have the bandwidth to provide support for git here.   I know git can be confusing for the new user, but to keep track of the code here it will be necessary to become more familiar with this tool.

I find the atlassin tutorials on using git to be very good and reference them myself when I can't remember how to do something:  https://www.atlassian.com/git/tutorials

To give you some basic info, git is telling you that you have made local changes to the code, and that you need to do something about that before it will let you overwrite those changes.

Usually in this case there are three options, in order of my personal preference:

1. You want to keep these changes, so you create a new branch, commit the changes to the branch, then switch to the other branch:
```
git checkout -b new-branch-name
git add <list of files that you want to keep track of>
git commit -m "descriptive commit message about what you changed"
git checkout altum-support
```

2. You want to keep the changes, but you don't want to track them in a branch, so you stash the changes in your local git stash (usually it's better to use a branch and then merge the branches):
```
git stash push
git checkout altum-support
```
when you want to re-integrate the changes (this will require merging, which can be a challenge!
```
git stash pop
```

3. You want to throw away the changes because you don't care about them (be careful there is no going back):

```
git reset --hard
git checkout altum-support
```

You can also do a combination of these.  You generally don't want to track every change to the .ipynb files, as these can become very large.  At the least you want to make sure to "reset and clear outputs" on the notebooks you have changed so no outputs/images are tracked.

So for example you can

```
git reset --hard *.ipynb
git stash push
git chekout altum-support
git stash pop  <-- this will cause a merge, which you may have to manually fix!
```

These are just suggestions and each of them may be wrong or break your code or checkout -- you can always make a copy of the folder you're running in before you do this until you're more comfortable with git.

andbrs, 2019-04-02:
hello, first of all thank you for the instructions served me a lot, now, I perform the tests and detect the panel in various positions. I think the bug is fixed.

[image]
[image]

closed 2019-04-09 by poynting