micasense/imageprocessing #35: California orchard data

Ciaran1981, 2019-03-14:
Which of the panel images in the large dataset in 'ImageSet Extended Examples notebook' are pre-flight calibration and post flight calibration? 

They are all in the same folder(000) so it is not clear what the arrangement is.

poynting, 2019-03-15:
Hi @Ciaran1981, the file referenced in the ImageSets page (https://micasense.github.io/imageprocessing/ImageSets.html) has two flight folders (000 and 001), but panel images are only present in 000.  For any RedEdge or Altum dataset, the images are numbered in the order they're taken per camera power on, so browsing those files will reveal there are only panel images taken before the flight.  Browsing the image metadata will also show this.  In this case the whole flight is only about 15 minutes, so a single panel image should be sufficient for calibration.

closed 2019-03-15 by poynting

Ciaran1981, 2019-03-15:
Hi Poynting,

Great, thanks for clarifying!

Ciaran