micasense/imageprocessing #36: imageutils mutliprocessing

Ciaran1981, 2019-03-15:
Hello,

I have encountered an error in

`imageutils.align_capture`

in line

`multiprocessing.set_start_method('spawn') `

where it returns a context already set error. I have solved this by  adding force=True to set_start_method (see below) and the function runs fine now:

`multiprocessing.set_start_method('spawn', force=True)`
 
Ciaran

poynting, 2019-03-18:
Does this only occur when that line of code is called in the same interpreter twice (e.g. in a loop)? If so, that's not a bug in the library, it's known to be the case.  As discussed in issue #18 set_spawn_method can only be called once in a program.

From what I've read, using `force=True` can cause other issues, the more accepted solution is just to catch the expected error, or, as described in the other issue, move it out to a higher level function.  You can also just check what the start_method is and call set accordingly.
```
if multiprocessing.get_start_method() != 'spawn':
    multiprocessing.set_start_method('spawn')
```
or
```
try:
    multiprocessing.set_start_method('spawn')
except RuntimeError:
    pass
```

poynting, 2019-03-23:
@Ciaran1981 which platform are you running on, and what python version?  We've seen a similar issue internally on MacOSX only, where the if statement option above fails because the start_method isn't 'spawn', but it has already been set by something else.

I'll give the `force=True` method a try and see what the results are.

Ciaran1981, 2019-03-27:
Hi poynting

Running on Ubuntu Bionic Beaver, py3.6 in a micasense conda env.

Ciaran

poynting, 2019-04-09:
Hi Ciaran,

Does the current version of the altum-support branch work out for you?  If you have some missing imports, check out the changes to the micasense_conda_env.yml file, there have been a couple of packages added.

poynting, 2019-04-10:
As this is working for me on fresh installs of mac/linux/windows, I'm going to close this and merge the altum-support.

If this is still a problem on the merged master, please let us know.

closed 2019-04-10 by poynting