micasense/imageprocessing #37: Applying warp to rest of image set

Ciaran1981, 2019-03-19:
More of a question than an issue. 

Which variable is required to apply a 'good' alignment transform to another capture and indeed the remaining dataset?

Is it simply a case of using 'warp_matrices' from a good aligment to the next capture?

poynting, 2019-03-20:
Yes, this is a little easier with the new altum-support branch. warp_matrices should be all that is required now. 

I'll post an example later this week days of using one alignment to stack and export a whole ImageSet to Tiff files and copy the metadata so that those files can be processed in Pix4D (or anything that can process TIFFs).

poynting, 2019-03-20:
I've added code to make this easier and added an example of this.  When run the Alignment notebook all the way through, the warp_matrices will be printed at the last step. These can then be set and used in the Batch Processing notebook to apply that warp across a whole imageset.  To see how this works or apply this idea to other work, follow the code through the process of saving a reflectance stack in the `capture.save_capture_as_reflectance_stack(...)` method.

It's important that the same warp method is used in both cases -- cv2.MOTION_AFFINE used to be the default, but now it's cv2.MOTION_HOMOGRAPHY, which we recommend using instead.  If you see `matix is not square` or `matrix sizes must match` errors it's probably a disconnect between these.

closed 2019-03-20 by poynting