micasense/imageprocessing #38: find_crop_bounds - Last 2 dimensions must be square

Ciaran1981, 2019-03-20:
Hello,

In the altum branch of the lib I get this error following this command: A bug perhaps?

 ```
cropped_dimensions, edges = imageutils.find_crop_bounds(imAl, 
                                                            warp_matrices,
                                                            warp_mode=cv2.MOTION_HOMOGRAPHY)
```
Here is the error feed culminating in 
`numpy.linalg.LinAlgError: Last 2 dimensions of the array must be square`
  ```

File "/home/ciaran/anaconda3/envs/micasense/lib/python3.6/site-packages/micasense/imageutils.py", line 302, in find_crop_bounds
    bounds = [get_inner_rect(s, a, d, c,warp_mode=warp_mode)[0] for s,a, d, c in zip(image_sizes,registration_transforms, lens_distortions, camera_matrices)]
  File "/home/ciaran/anaconda3/envs/micasense/lib/python3.6/site-packages/micasense/imageutils.py", line 302, in <listcomp>
    bounds = [get_inner_rect(s, a, d, c,warp_mode=warp_mode)[0] for s,a, d, c in zip(image_sizes,registration_transforms, lens_distortions, camera_matrices)]
  File "/home/ciaran/anaconda3/envs/micasense/lib/python3.6/site-packages/micasense/imageutils.py", line 321, in get_inner_rect
    left_map = map_points(left_edge, image_size, affine, distortion_coeffs, camera_matrix,warp_mode=warp_mode)
  File "/home/ciaran/anaconda3/envs/micasense/lib/python3.6/site-packages/micasense/imageutils.py", line 391, in map_points
    new_pts =cv2.perspectiveTransform(new_pts,np.linalg.inv(warpMatrix).astype(np.float32))
  File "/home/ciaran/anaconda3/envs/micasense/lib/python3.6/site-packages/numpy/linalg/linalg.py", line 546, in inv
    _assertNdSquareness(a)
  File "/home/ciaran/anaconda3/envs/micasense/lib/python3.6/site-packages/numpy/linalg/linalg.py", line 213, in _assertNdSquareness
    raise LinAlgError('Last 2 dimensions of the array must be square')
numpy.linalg.LinAlgError: Last 2 dimensions of the array must be square

```

poynting, 2019-03-20:
Please be more specific about what you're doing when this bug occurs. Are you running the code from one of the supplied notebooks? Are you running your own script?

One way I can imagin this happening is when using cv2.MOTION_AFFINE transform in a function that's expecting a cv2.MOTION_HOMOGRAPHY.

Ciaran1981, 2019-03-22:
Hi Again,

Yes - it was indeed a mismatch of algorithms in my script. Apologies for this! 

Many thanks for your help again with applying warp-matrices to entire datasets.

closed 2019-03-23 by poynting