micasense/imageprocessing #42: Improve imageset loading time

poynting, 2019-03-29:
Currently we load the images into an imageset from a list of images or a directory. In this process, the image data itself isn't loaded, but the metadata is read via exiftool.  This process is very slow.

I've tried using one exiftool object and loading all images from that, which should improve the setup/teardown efficiency of exiftool, but this didn't impact the loading speed much.  It's possible I implemented this incorrectly; it's worth exploring.  

The suggested approach would be to use the exiftool -csv option to read the exif data from a set of images, then build those image objects from the data in the csv. The exiftool process to save metadata to a CSV is relatively quick, running in seconds instead of minutes on a set of a few thousand image files.

swawa, 2019-04-01:
Hi poynting,

(First: I am using the Altum camera)
I try to follow up with the speed enhancement. I want to use the timeit to search for bottlenecks in the existing procedures. For that I want to run them "as they are" at the moment. I receive an error by image.py when running the image set reading function:
[image]

General ideas for speeding things up:
-using all available processors and calculate in parallel like with Process Pool
Swawa

poynting, 2019-04-01:
A couple things:

The code as currently committed doesn't have a return statement on that
line, so I maybe you have made a change that is breaking? Do all your tests
pass with a clean checkout?

https://github.com/micasense/imageprocessing/blob/altum-support/micasense/image.py#L402

Second, the imageset loading function already already uses
multiprocessing.  I believe the bottleneck is calling exiftool through
pyexiftool. I think we need to move to use a single exiftool call to create
a csv and then load the images as Python objects from that metadata file.
As described above, I tried re-using a single exiftool object but that
didn't speed things up. I may have done that incorrectly, it may be worth
trying to repeat. I just created one global exiftool object in ImageSet and
then passed it down into the image creation function.

poynting, 2019-04-01:
Here's a quick way to profile that verifies what I expected:

running `ipython` on the anaconda prompt:
```
import micasense.image as image
import cProfile
cProfile.run('image.Image("data/0000SET/000/IMG_0000_1.tif"))
```
results in the following stats.  Note that most of the calls are <1ms (they don't register) but the nt.open call within the pyexiftool exec takes ~0.22 seconds and runs twice.
```
Mon Apr  1 14:27:10 2019    imgstats

         440 function calls in 0.448 seconds

   Ordered by: internal time

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
        2    0.411    0.205    0.411    0.205 {built-in method nt.read}
        5    0.028    0.006    0.028    0.006 {method 'acquire' of '_thread.lock' objects}
        1    0.005    0.005    0.005    0.005 {built-in method _winapi.CreateProcess}
        1    0.002    0.002    0.002    0.002 {built-in method _winapi.WaitForSingleObject}
        7    0.000    0.000    0.000    0.000 {built-in method _winapi.CloseHandle}
        1    0.000    0.000    0.000    0.000 {built-in method _thread.start_new_thread}
        1    0.000    0.000    0.000    0.000 {built-in method nt.stat}
        1    0.000    0.000    0.000    0.000 \envs\micasense\lib\json\decoder.py:343(raw_decode)
        3    0.000    0.000    0.000    0.000 {built-in method io.open}
        1    0.000    0.000    0.448    0.448 micasense\image.py:64(__init__)
        1    0.000    0.000    0.005    0.005 \envs\micasense\lib\site-packages\exiftool.py:158(start)
        1    0.000    0.000    0.411    0.411 \envs\micasense\lib\site-packages\exiftool.py:200(execute)
       68    0.000    0.000    0.000    0.000 micasense\metadata.py:52(get_item)
        1    0.000    0.000    0.000    0.000 \envs\micasense\lib\_strptime.py:318(_strptime)
        2    0.000    0.000    0.000    0.000 {built-in method _winapi.CreatePipe}
        1    0.000    0.000    0.005    0.005 \envs\micasense\lib\subprocess.py:1101(_execute_child)
        1    0.000    0.000    0.000    0.000 \envs\micasense\lib\subprocess.py:485(list2cmdline)
        2    0.000    0.000    0.031    0.015 \envs\micasense\lib\site-packages\exiftool.py:177(terminate)
        1    0.000    0.000    0.448    0.448 {built-in method builtins.exec}
        1    0.000    0.000    0.029    0.029 \envs\micasense\lib\subprocess.py:1236(_communicate)
        1    0.000    0.000    0.005    0.005 \envs\micasense\lib\subprocess.py:650(__init__)
        1    0.000    0.000    0.447    0.447 micasense\metadata.py:36(__init__)
        1    0.000    0.000    0.000    0.000 micasense\metadata.py:118(utc_time)
        1    0.000    0.000    0.448    0.448 <string>:1(<module>)
        1    0.000    0.000    0.001    0.001 \envs\micasense\lib\threading.py:829(start)
        1    0.000    0.000    0.411    0.411 \envs\micasense\lib\site-packages\exiftool.py:229(execute_json)
        2    0.000    0.000    0.000    0.000 {method 'flush' of '_io.BufferedWriter' objects}
        1    0.000    0.000    0.000    0.000 \envs\micasense\lib\subprocess.py:1007(_get_handles)
        1    0.000    0.000    0.000    0.000 {built-in method _locale._getdefaultlocale}
        1    0.000    0.000    0.000    0.000 \envs\micasense\lib\ntpath.py:450(normpath)
        1    0.000    0.000    0.000    0.000 {method 'close' of '_io.BufferedWriter' objects}
        1    0.000    0.000    0.000    0.000 \envs\micasense\lib\threading.py:758(__init__)
        1    0.000    0.000    0.002    0.002 \envs\micasense\lib\subprocess.py:1215(_wait)
        1    0.000    0.000    0.031    0.031 \envs\micasense\lib\subprocess.py:889(communicate)
       11    0.000    0.000    0.000    0.000 \envs\micasense\lib\subprocess.py:202(Close)
        3    0.000    0.000    0.000    0.000 {method 'match' of 're.Pattern' objects}
        1    0.000    0.000    0.000    0.000 \envs\micasense\lib\threading.py:264(wait)
        4    0.000    0.000    0.000    0.000 micasense\metadata.py:77(size)
        1    0.000    0.000    0.000    0.000 \envs\micasense\lib\threading.py:216(__init__)
        1    0.000    0.000    0.000    0.000 \envs\micasense\lib\threading.py:534(wait)
        1    0.000    0.000    0.000    0.000 \envs\micasense\lib\json\decoder.py:332(decode)
        3    0.000    0.000    0.000    0.000 {built-in method _thread.allocate_lock}
        1    0.000    0.000    0.000    0.000 \envs\micasense\lib\threading.py:968(_stop)
       31    0.000    0.000    0.000    0.000 {built-in method builtins.isinstance}
        1    0.000    0.000    0.028    0.028 \envs\micasense\lib\threading.py:1000(join)
        1    0.000    0.000    0.000    0.000 \envs\micasense\lib\json\__init__.py:299(loads)
        1    0.000    0.000    0.028    0.028 \envs\micasense\lib\threading.py:1038(_wait_for_tstate_lock)
        3    0.000    0.000    0.000    0.000 \envs\micasense\lib\subprocess.py:1079(_make_inheritable)
        1    0.000    0.000    0.000    0.000 micasense\metadata.py:135(dls_pose)
        1    0.000    0.000    0.000    0.000 {built-in method _locale.setlocale}
        3    0.000    0.000    0.000    0.000 {built-in method _winapi.DuplicateHandle}
        1    0.000    0.000    0.000    0.000 \envs\micasense\lib\_weakrefset.py:38(_remove)
        1    0.000    0.000    0.000    0.000 \envs\micasense\lib\_strptime.py:574(_strptime_datetime)
[... 104 lines cut ...]
        1    0.000    0.000    0.000    0.000 {method 'add' of 'set' objects}
[... cut ...]
        1    0.000    0.000    0.000    0.000 {method 'disable' of '_lsprof.Profiler' objects}
        2    0.000    0.000    0.000    0.000 {method 'toordinal' of 'datetime.date' objects}
        1    0.000    0.000    0.000    0.000 {built-in method msvcrt.get_osfhandle}
        3    0.000    0.000    0.000    0.000 \envs\micasense\lib\threading.py:507(is_set)
        2    0.000    0.000    0.000    0.000 {method 'extend' of 'list' objects}
        1    0.000    0.000    0.000    0.000 {method 'weekday' of 'datetime.date' objects}
```

poynting, 2019-04-01:
Hi @swawa:

Turns out I did have a mistake in my earlier test.  I've reduced the example imageset to 26 seconds using the batch mode on exiftool vs about 3 minutes 30 seconds using the previous multiprocessing mode.

I should have a patch out in a couple hours.

poynting, 2019-04-01:
Committed an approximately 7x improvement in speed in 28f114c8c0a3fd28787b7664b86eb844e44cb12a

Imageset loads the metadata from 1675 image files in approx 25-30 seconds.  This is within a few seconds of the 22 seconds it takes to run the command `exiftool -csv -r imageDir > images.csv`

It's possible this can be improved further, it's back to single-threaded, but removes the per-image overhead of creating the exiftool process. The problem I had with multithreading this was that I can't come up with a way to pass the common complex exiftool object to the child processes, since it's not pickleable.  Creating a global object is bad form, but also didn't seem to work.

I think we can call this done for now if you agree @swawa

loicdtx, 2019-04-02:
This fails on linux if `exiftoolpath` environment variable is not set (Line `exiftool_path = os.path.normpath(os.environ.get('exiftoolpath'))` throws `TypeError: expected str, bytes or os.PathLike object, not NoneType`).
I applied a dirty fix on my fork (see https://github.com/loicdtx/imageprocessing/commit/359bcf2a1b906e9a8bd4f73ca1752dd87344364e ) I'm sure you'll find a more elegant way to deal with the issue :)

poynting, 2019-04-02:
> I applied a dirty fix on my fork (see loicdtx@359bcf2

Thanks, I expected that call to just return None. I'll fix this.

poynting, 2019-04-02:
I think the path issue should be fixed by that merge @loicdtx -- please let me know if it doesn't fix it for you.  Your solution was just as valid; my embedded C roots just lead me to avoid exceptions if possible.

swawa, 2019-04-03:
Hi Poynting,
There was somehow a formatting error within some of my files. I pulled the whole branch again and now it seems to work. Sorry for the confusion! For me it is fine now. In my eyes, the speed is good for that amount of images.

closed 2019-04-09 by poynting