micasense/imageprocessing #47: Altum irradiance values

tuncaemre, 2019-04-02:
Hello, I try to process Altum images, when I study on this images I realize that altum's Spectral Irradiance value 104.20955143678957 which is used for irradiance correction. But in rededge spectral irradiance value like a 1.084824800491333. Nearly 100 times lower. Is there any problem when we calculate reflectance values of image set using dls+panel together.

Best

poynting, 2019-04-03:
Hi @tuncaemre:

For cameras with DLS2, the scale factor for all irradiance tags has changed. For DLS1, the units were the SI standard W/m^2/nm.  For current DLS2 data, the units are uW/cm^2/nm.  This results in a scale difference of 100 between the two unit sets.  Altum or RedEdge doesn't matter - the difference is DLS1 vs DLS2.  This was ultimately an unintentional change, as we strongly prefer SI units, but we had some calibration software and validation software that was canceling this change so we didn't notice until recently.  The altum-support branch of this code handles the change correctly.  Here's how you should handle this (and I'll add this more complete handling to this library as well):

If the image metadata has a horizontal_irradiance tag, but no irradiace_scale_to_si tag, you should multiply the spectral_irradiance and horizontal irradiance by 0.01 (divide by 100) to get to W/m2/nm.  If there is an irradiance_scale_to_si tag, then you should scale the data by that tag.  We haven't released a firmware that has the irradiance_scale_to_si tag, but will likely do so in the future.

In metadata.py:
```python
def irradiance_scale_factor(self):
    if self.get_item('XMP:IrradianceScaleToSIUnits') is not None:
         # the metadata contains the scale
        scale_factor = self.__float_or_zero(self.get_item('XMP:IrradianceScaleToSIUnits'))
    elif self.get_item('XMP:HorizontalIrradiance') is not None: 
        # DLS2 but the metadata is missing the scale, assume 0.01
        scale_factor = 0.01
    else:
        # DLS1, so we use a scale of 1.0
        scale_factor = 1.0
    return scale_factor
```

If you're using both panels and DLS, this difference will just 'fall out' of calibrating with both -- your correction panel for the DLS when using the panel will just be closer to 100 (or 0.01) than 1.0.

tuncaemre, 2019-04-04:
Hi poynting,
when I calculate sensor_irradiance (sensor_irradiance = img.dls_irradiance / fresnel_correction), this formula calculate sensor irradiance using Spectral Irradiance rather than Horizontal Irradiance.
maybe it should be changed.

best.

poynting, 2019-04-04:
I'm sorry I don't understand what you mean. Can you provide a code sample of what you're doing, and a code sample of what you expect?

poynting, 2019-04-05:
@tuncaemre : I'm not sure exactly what you were looking for, but take a look at the [most recent commit ](https://github.com/micasense/imageprocessing/commit/779b3592e933c0feeaf5e63f057da57e0e85451d) where I refactored the dls interface substantially.

For most purposes, Image.horizontal_irradiance() should be what you want. It should handle both DLS1 and DLS2, as well as some DLS2 errata on specific firmware versions where horizontal_irradiance was invalid (but the other irradiance tags were OK).

tuncaemre, 2019-04-05:
Sorry for late response,
I mean that in MicaSense Image Processing Tutorial 3 for Altum to obtain reflectance images, in Correcting DLS readings for orientations section using this code:
# Since the diffuser reflects more light at shallow angles than at steep angles,
# we compute a correction for this
fresnel_correction = dls.fresnel(sun_sensor_angle)
# Now we can correct the raw DLS readings and compute the irradiance on level ground
dls_irradiances = []
center_wavelengths = []
for img in cap.images:
    dir_dif_ratio = 6.0
    percent_diffuse = 1.0/dir_dif_ratio
    # measured Irradiance / fresnelCorrection
    sensor_irradiance = img.dls_irradiance / fresnel_correction
    untilted_direct_irr = sensor_irradiance / (percent_diffuse + np.cos(sun_sensor_angle))
    # compute irradiance on the ground using the solar altitude angle
    dls_irr = untilted_direct_irr * (percent_diffuse + np.sin(solar_elevation))
    dls_irradiances.append(dls_irr)
    center_wavelengths.append(img.center_wavelength)

import matplotlib.pyplot as plt
plt.scatter(center_wavelengths,dls_irradiances)
plt.xlabel('Wavelength (nm)')
plt.ylabel('Irradiance ($W/m^2/nm$)')
plt.show();
cap.plot_undistorted_reflectance(dls_irradiances)

when we calculate sensor_irradiance (sensor_irradiance = img.dls_irradiance / fresnel_correction) img.dls_irradiance using Spectral Irradiance exif value of image. But I think that it should be ,
sensor_irradiance = img.horizontal_irradiance / fresnel_correction

My other confuse is:
after make correction which is shown above code as a "dls_irr", irradiance values shown as a  
54.0298357727151;
47.77953992378548;
40.55560558151846;
26.052343220364754;
31.73878153528794

these values are high or not?

poynting, 2019-04-05:
Tutorial #3 walks through correction of DLS1 data for the example RedEdge images.  DLS2 data (any data with Altum or RedEdge-MX) must be handled differently.

See the [comment above](https://github.com/micasense/imageprocessing/issues/47#issuecomment-479282542) about the units.  **The units have changed from DLS1 to DLS2**.  This is the direct answer to your question about the "high values."

In addition, DLS2 data is handled differently, as the camera system has already estimated most of the parameters for you. I'll update the tutorial, but for example code that handles both DLS correctly, see https://github.com/micasense/imageprocessing/blob/altum-support/micasense/image.py#L112

In short, all of the angular corrections required for DLS1 are **not** required for DLS2.  If you're using the most recent firmware (>=1.2.3 for Altum), all you will need to do is extract the HorizontalIrradiance tag and scale it properly. If you're using older Altum firmware (<1.2.3), you will need to extract the scattered and direct irradiances from the metadata, and then use them along with the sun angles to calculate the horizontal irradiance.  The code may appear complex because it handles all of these cases.

tuncaemre, 2019-04-05:
ok, thank you so much clarify it.

closed 2019-04-09 by poynting