micasense/imageprocessing #50: export image with ndvi or ndre in .TIFF

andbrs, 2019-04-05:
hello, I would like to know how to export an NDVI image. the .tiff file for analysis in QGIS or PIX4D, because, in the alignment tutorial leaves only one image with all the bands together. Thanks

poynting, 2019-04-05:
To analyze a single image, just open the "stack" Tiff image  in QGIS.  The
you can use the raster calculator to calculate anything you like. however a
single image will not be geofectified.

If want to run this code and generate maps, run the Batch processing
notebook to create a whole set of Tiff files. As long as the notebook runs
all the way through, it will give you a folder full of Tiff files which
have the correct GPS coordinates and time. Then you can load these into
Pix4D to match them together and create a map and 3D model.

Then you will have a full map as a big Tiff file, and you can use QGIS or
ArcGis to open it and perform analysis.

closed 2019-04-05 by poynting

andbrs, 2019-04-06:
I understand, so I must conclude that there is no way to extract the image with pseudocolor from any folder of the project?

poynting, 2019-04-06:
Of course there is a way; there is an example of doing this already in the
Alignment-RigRelatives tutorial. That tutorial shows how to render NDVI and
NDRE, and the last (thermal) image in that tutorial is saved after
rendering.

Please read that code, run it, and then take a try at modifying it to do
what you need, and if you have specific questions let us know.

andbrs, 2019-04-11:
I managed to save these images and is in .tiff format, I must test if for example with qgis, one more question the program works with Jupyter but if I try to make it work with SPYDER ide in the alignment of images remains as in a bug, why does this happen?

poynting, 2019-04-12:
That's great!  I have no idea why Spyder doesn't work, it's not a tool that
I use.

martinchez, 2019-09-01:
how did you  save or export the ndvi  image in .tiff file format

poynting, 2019-09-04:
@martinchez  See the "Stack Export" section in the Alignment notebook.  You'll want to follow along with that code, but change a few things.  For example you'll need to change

```outRaster = driver.Create(filename+".tiff", cols, rows, im_aligned.shape[2], gdal.GDT_UInt16)``` 

to use one band instead of `im_aligned.shape[2]` bands and change the type to the `gdal.GDT_Float` type, and then write the data you would like (`ndvi`) to the `outband` using for example

```outband.WriteArray(ndvi)```

So most of the code of that block goes away to create a new output block, and then you'll need to call this new code after you compute the ndvi matrix.