micasense/imageprocessing #51: Alignment is looping - any glue?

swawa, 2019-04-08:
Hi,
I am processing the alignment of one Altum capture like described in the alignment tutorial (Windows machine).
In my processing, the opening works well as well as the irradiance and undistorted reflectance images.
There is always hangig up in a loop with the following instruction (Part [3]):
warp_matrices, alignment_pairs = imageutils.align_capture(capture, max_iterations=1)
-yes, even if I set the iterations only to one.

There were a lot of exiftool windows opening and closing one time. Usually this seems to be a hidden process. But it might be a hint...
Any glue why this might happen? (Everything is updated to the newest version.)

Best wishes,
Swawa

poynting, 2019-04-08:
It would probably help to provide more context: what code are you running (are you running your own code "like described" - if so provide that code) or are you running the actual tutorial notebook exactly?  What OS? What's your pip freeze look like?

I'm probably not going to be able to help you - these multiprocessing quirks across different platforms seem to be a quite deep rabbit hole that I'm not sure what else to do with.  Maybe someone else here has some ideas based on the further information you provide.

poynting, 2019-04-08:
Also, you can just call align_capture(..., multithreaded=False) to bypass the multiprocessing stuff entirely if it isn't working for you. The only difference should be the amount of time it takes to complete.

swawa, 2019-04-09:
Hi Poynting,

thanks for the answer. It is still Windos 7. And the code was untouched like in the "Alingment" tutorial. The hint with  align_capture(..., multithreaded=False) solved the problem. That is really interesting...  Ashame. The multprocessing was my own suggestion a few days ago... ... It is running through now, cool! thanks. :-)

poynting, 2019-04-09:
The multiprocessing for image alignment has been there for quite a long time; I think you were having issues with the speed of imageset loading and wanting to use multiprocessing in issue #42.  

In it's current state, the multiprocessing-enabled image alignment works fine for me with fresh environment installations on modern versions of Windows (10), Mac, and Linux, which is why I'm not sure how else to help.  It would be great if someone else has an idea for a solution.  It seems there are a lot of multiprocessing quirks across the different platforms.

swawa, 2019-04-09:
Maybe it isnt really worth the effort... as Windows 7 is disappearing.... I am looking forward that our oparting system will be exchanged somewhen :-)

Now the croping is also stating an error:

  File "[.....]imageprocessingAltum/Capture_compute.py", line 142, in <module>
    cam_mats)

TypeError: find_crop_bounds() takes from 2 to 3 positional arguments but 4 were given
(the given ones are:
cropped_dimensions = imageutils.find_crop_bounds(capture.images[0].size(), 
                                                 warp_matrices, 
                                                 dist_coeffs, 
                                                 cam_mats)

Why that? Shall I throw away my operating system ASAP? ;-)

poynting, 2019-04-09:
It looks like maybe you're running your own code? I suspect you will need
to update some of the calls as some of the function arguments, especially
around the alignment code, have changed.

swawa, 2019-04-09:
hm... I just copied the code out of the website. But I see, it is different today. I'll check that..

swawa, 2019-04-10:
interstingly, the Passive Image Alignment tutorial works well with the passive alignment

swawa, 2019-04-25:
With the new Alignment tutorial https://micasense.github.io/imageprocessing/Alignment.html
it is working now....

swawa, 2019-04-26:
From the same tuorial I get a "zero" NDVI-histogram for the NDVI-masking. There is this hint below the histogram image:
numpy\lib\function_base.py:3826: RuntimeWarning: Invalid value encountered in percentile interpolation=interpolation
Each single channel histogram looks good. any idea why the part

> # Compute and display a histogram
> ndvi_hist_min = np.min(ndvi)
> ndvi_hist_max = np.max(ndvi)
> fig, axis = plt.subplots(1, 1, figsize=(10,4))
> axis.hist(ndvi.ravel(), bins=512, range=(ndvi_hist_min, ndvi_hist_max))
> 

shows a zero-line histogram?

poynting, 2019-04-27:
@swawa I don't see those results when running the default tutorial with the included data; I just did a full update/reinstall on a new windows machine and I get the same results as the link above for the Alignment tutorial.

Are you using your own data? Have you tried commenting/removing or modifying the values of the masking lines just above that section?

swawa, 2019-04-29:
Hi poynting,

thanks. yes, I run it with my own data. The sensor was also updated to the latest firmware version before going through the tutorial but no new machine/windows setup and installing of the branch. It should not necessary each time we chance the code, right?
I also am not authorized to set up a completely new machine as our IT department is the only one who is allowed to do so. 
I tried to comment, and with different values. No change. And also reinstalling anaconda and the micasense environment. I will try to make a new set of images. maybe something went wrong while taking the other images...?

poynting, 2019-04-29:
It's usually not necessary to reinstall the environment, but can be required when the micasense_conda_env.yml file changes due to new/different libraries being used.  As libraries are updated and changed by those package maintainers, we sometimes have to pin to an older version to ensure compatibility.  In an ideal world we would pin all versions, but doing so early on led to an inability to create environments across platforms.

As a note, the altum-support branch has been merged into master, and now master has been updated with bugfixes and is the currently active branch. 

In any case, it won't be possible to help further without trying your data. If you can share at least one panel image and one flight image I can take a look.

moghi005, 2019-06-17:
We have RedEdge with the software version V.5.1.8 - however, the metadata doesn't have RigRelatives tag, while in tutorial it was mentioned that any version above 3.2.0 should have. 

I need this for alignment. Could you please let me know why we don't have it? And how we can have it?

poynting, 2019-06-17:
@moghi005 please contact MicaSense support for that issue and they can help you out; there was a bug in some firmware versions depending on your camera type and you will need to update to 5.1.12.  Support can help you work through it and verify the fix.

Alignment can be done successfully without rig relatives; set `pyramid_levels = None` in the third cell in the Alignment notebook and use an image with a lot of clear edges in every band.

poynting, 2020-01-30:
@swawa is it possible the issue with alignment looping has been that the execution is happening outside of a notebook and it isn't wrapped in an `if main` block?

```
if __name__ == '__main__':
  #your code
```

https://github.com/micasense/imageprocessing/issues/104#issuecomment-578950302
(fully executable example a few comments down from that)

poynting, 2020-04-27:
Closing due to inactivity. The above comment should fix this issue

closed 2020-04-27 by poynting