micasense/imageprocessing #52: Error when testing installation

smittybombs, 2019-04-08:
After going through the installation process, I tried running the test snippet in Jupyter and I get an error. The error occurs when it tries to run Mapboxgl. The installation seemed to go ok. I tried importing them one by one as well and got the same error when I ran the Mapboxgl. I tried doing a pip install for Mapboxgl and still get the error. I am very new to Micasense, multispectral imagery and Python, so forgive me if I am doing something dumb! I have searched online and in the Micasense Knowledgebase and have not found anything helpful. Thanks in advance!

---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-1-3af66e9ef526> in <module>
      6 import pyzbar.pyzbar as pyzbar
      7 import matplotlib.pyplot as plt
----> 8 import mapboxgl
      9 
     10 print()

~\Anaconda3\envs\micasense\lib\site-packages\mapboxgl\__init__.py in <module>
----> 1 from .viz import CircleViz, GraduatedCircleViz, HeatmapViz, ClusteredCircleViz, ImageViz, RasterTilesViz, ChoroplethViz, LinestringViz
      2 
      3 __version__ = "0.10.1"
      4 __all__ = ['CircleViz', 'GraduatedCircleViz', 'HeatmapViz', 'ClusteredCircleViz', 'ImageViz', 'RasterTilesViz', 'ChoroplethViz', 'LinestringViz']

~\Anaconda3\envs\micasense\lib\site-packages\mapboxgl\viz.py in <module>
      6 
      7 import numpy
----> 8 import requests
      9 
     10 from mapboxgl.errors import TokenError

ModuleNotFoundError: No module named 'requests'

poynting, 2019-04-08:
This is fixed in the current development branch, `altum-support`. You can also fix by running conda install requests in your activated environment, or by down-grading mapbox-gl to a version that doesn't require requests (<0.7.x I think)

poynting, 2019-04-09:
Fixed in commit 133b04, will be merged to master in the next couple of days.

closed 2019-04-09 by poynting