micasense/imageprocessing #55: non-alignment of the rededge and NIR bands

andbrs, 2019-04-12:
I am testing the results of the multispectral image with combined bands in matlab, and at the moment of testing this I can see that the bands R,G,B if they are aligned (figure 2) on the other hand if we analyze the bands NIR and RedEdge we can see that they are not alienated (figure 1) from each other therefore, they have not been aligned (geometric correction).
In order to determine if they are aligned I use matlab with this command: imshowpair(NIR,rededge,'blend');
-Any idea why this happens? or is it normal? 
-Doesn't it affect the analysis of the vegetative index? 

[image]
[image]

poynting, 2019-04-12:
I'm sorry but in the previous question where I processed the data for you
they were aligned as is shown in the notebook I sent. Maybe you can try
those warp matrices and see what results you get.

Aligning data on the ground like this has its own challenge, and isnt the
point of this tutorial, but I've already provided code and examples to help
with this case.

andbrs, 2019-04-12:
these images are the results of the example notebook which sent me the image called "bgrne" which is the 5 bands combined, we used the matlab software to see each band separately and let you see which was not aligned comparing them with each other. I didn't really understand the last answer very well, I could explain. Thank you very much

poynting, 2019-04-12:
This library is designed for images taken from drones, where the subject is very far away. It is not designed to work with images taken from only a meter away or less.  However, I've already provided some customized support to help you with this specific question. In that response, I provided a notebook which shows that alignment is possible if you use a specific set of parameters and modify the library slightly.  

https://github.com/micasense/imageprocessing/issues/45#issuecomment-479177801

I'm very sorry but I can't provide any more support on this subject. If you have some images taken from the air, pointing straight down, more than 20m away from the subject, this library may work much better for you.

closed 2019-04-12 by poynting