micasense/imageprocessing #58: JSONDecodeError: Expecting value: line 1 column 1 (char 0)

niuhaoyu16, 2019-05-13:
Successfully imported all required libraries.

Successfully executed exiftool.

Successful for the above. But got an error for the below

from micasense.image import Image
imagePath = os.path.join('.','data','0000SET','000')
imageName = glob.glob(os.path.join(imagePath,'IMG_0000_1.tif'))[0]

img = Image(imageName)

JSONDecodeError                           Traceback (most recent call last)
<ipython-input-3-9c25cd451476> in <module>
      3 imageName = glob.glob(os.path.join(imagePath,'IMG_0000_1.tif'))[0]
      4 
----> 5 img = Image(imageName)

C:\Windows\System32\imageprocessing\micasense\image.py in __init__(self, image_path, exiftool_obj)
     67             raise IOError("Provided path is not a file: {}".format(image_path))
     68         self.path = image_path
---> 69         self.meta = metadata.Metadata(self.path, exiftool_obj=exiftool_obj)
     70 
     71         if self.meta.band_name() is None:

C:\Windows\System32\imageprocessing\micasense\metadata.py in __init__(self, filename, exiftoolPath, exiftool_obj)
     47             raise IOError("Input path is not a file")
     48         with exiftool.ExifTool(self.exiftoolPath) as exift:
---> 49             self.exif = exift.get_metadata(filename)
     50 
     51     def get_all(self):

~\Anaconda3\envs\micasense\lib\site-packages\exiftool.py in get_metadata(self, filename)
    266         documentation of :py:meth:`execute_json()`.
    267         """
--> 268         return self.execute_json(filename)[0]
    269 
    270     def get_tags_batch(self, tags, filenames):

~\Anaconda3\envs\micasense\lib\site-packages\exiftool.py in execute_json(self, *params)
    250         """
    251         params = map(fsencode, params)
--> 252         return json.loads(self.execute(b"-j", *params).decode("utf-8"))
    253 
    254     def get_metadata_batch(self, filenames):

~\Anaconda3\envs\micasense\lib\json\__init__.py in loads(s, encoding, cls, object_hook, parse_float, parse_int, parse_constant, object_pairs_hook, **kw)
    346             parse_int is None and parse_float is None and
    347             parse_constant is None and object_pairs_hook is None and not kw):
--> 348         return _default_decoder.decode(s)
    349     if cls is None:
    350         cls = JSONDecoder

~\Anaconda3\envs\micasense\lib\json\decoder.py in decode(self, s, _w)
    335 
    336         """
--> 337         obj, end = self.raw_decode(s, idx=_w(s, 0).end())
    338         end = _w(s, end).end()
    339         if end != len(s):

~\Anaconda3\envs\micasense\lib\json\decoder.py in raw_decode(self, s, idx)
    353             obj, end = self.scan_once(s, idx)
    354         except StopIteration as err:
--> 355             raise JSONDecodeError("Expecting value", s, err.value) from None
    356         return obj, end

JSONDecodeError: Expecting value: line 1 column 1 (char 0)

poynting, 2019-05-13:
Is the file you're trying to open there and populated?

We recently moved all of the tutorial data into git-lfs, this makes the initial checkout smaller but you may need to do a secondary operation that I didn't document:

`git lfs fetch`

git lfs should have been installed as part of the conda installation; if not it can be installed manually here https://git-lfs.github.com/

Please let me know if that works

niuhaoyu16, 2019-05-14:
I still have the same problem after git lfs fetch successful.

The problem is because of this following line:

img = Image(imageName)

poynting, 2019-05-14:
I'm not sure how I can help you without more information. Can you verify
that you can open and view the image files in the data directory and that
that appear as they do in the documentation? I still suspect there is an
issue with the LFS download.

niuhaoyu16, 2019-05-14:
No, as you mentioned, the image can't be open.

niuhaoyu16, 2019-05-14:
Even if i change the image to my own .tif images, i still got the same error

poynting, 2019-05-14:
On the one hand you say "no, as you mentioned, the image can't open" and on the other hand you say if you change to your own image you get the same error.  

This error looks like exiftool is trying to open a file that isn't actually an image file.  Without using the library, can you open the image in a normal file browser? Can you browse your file browser to the `./data/000SET/000/` directory and double click on `IMG_0000_1.tif`?  Does the file open and show the image you expect?

From that same directory, can you open a command line and run `exiftool IMG_0000_1.tif`? Does that show an error?

niuhaoyu16, 2019-05-14:
Hi, thanks for the reply. I can't open the files in the data, though they are .tif. 

[image]

[image]

niuhaoyu16, 2019-05-14:
And, am i running the  git lfs fetch in the right way? Thanks
[image]

poynting, 2019-05-14:
I think this means that the files aren't really there - they're just git lfs file pointers.

This is a challenge with `git lfs` -- if you don't have it installed on the first checkout things can get into a strange state.

Something to try:

1. In the checkout directory run `git lfs install`
2. In the checkout directory run `git fetch`

If this doesn't take at least a few seconds while downloading, it's probably not working.  

If that doesn't work, wipe out your download directory (you won't have to re-create the python environment):

1. Make sure you have git lfs installed (run `git lfs` and you should see the help text)
2. Delete the cloned directory 
3. Run `git clone` again

This time the lfs objects should get pulled.

Let me know if this works and I'll update the documentation.

poynting, 2019-09-04:
@niuhaoyu16 can you confirm if using the steps above fixed your issue?

poynting, 2019-10-04:
Closing due to inactivity / lack of response.  The issue was re-created when git-lfs was not installed when cloning the repository; it's important to have git lfs installed before cloning for the lfs files to be pulled correctly.

closed 2019-10-04 by poynting