micasense/imageprocessing #60: Update readme.md to provide instructions for installing as a module

poynting, 2019-05-14:
Let's update the documentation to capture that the library can be installed as a module.  We should include an explanation of the `pip install -e .` or `python setup.py develop` syntax to capture that many developers may want to install in place so they can continue to update and develop the library code.

mmcph005, 2019-06-30:
I usually use jupyter lab instead of the notebook and ran into the issue of not finding micasense module when 
`import micasense`

However, I saw that it successfully imported micasense when I used it in the notebook. I also saw the micasense folder and files in there, so I assumed the micasense library needed to be added as a module in conda.
I then found some tidbits here and [here](https://stackoverflow.com/questions/37006114/anaconda-permanently-include-external-packages-like-in-pythonpath) which led me to the following steps:

1. `cd imageprocessing`
2. `conda develop micasense`
3. `python3 setup.py install`
4. `jupyter lab .`
5. `import micasense`

Now it works.

poynting, 2019-11-20:
Those are good tips, I haven't used lab yet.  One thing I'll add is that for users that want to modify library code, 3 can be replaced with
```
python3 setup.py develop
```
which allows the user to continue editing the installed library code without re-installing.  When done hacking on the code, the `python3 setup.py install` option can be used.

There is also some commentary [here](https://stackoverflow.com/questions/19048732/python-setup-py-develop-vs-install) about using `pip -e` instead of setup.py as a preferred alternative due to how it handles dependencies.

kvram55, 2019-11-28:
How can I install module named 'micasense'. 
I am getting error "ModuleNotFoundError: No module named 'micasense'

poynting, 2021-04-19:
The library supports installing as a module. Follow the setup instructions, and when done, run

```
python setup.py install
```

from the checkout directory

closed 2021-04-19 by poynting