micasense/imageprocessing #61: How to ignore DLS data in batch processing

esteps, 2019-05-16:
Hello,

I'm trying to figure out how to process a set of images where I correct to reflectance using only the panel (not the DLS data). My images have DLS data but I want to ignore it because the DLS was not operating properly.  Is this possible using the Batch Processing script? The reason I'm confused is because it seems like the script only calculates radiance if the DLS is ignored.

Any feedback is much appreciated!

poynting, 2019-05-16:
That's what the batch processing should do if you provide a panel capture like the example does.

The way it should work if you provide a panel capture, only the (single) panel irradiance is used in `create_aligned_capture`.  You need to be using a newer panel with the information embedded in the QR code (any panel with a table of values printed on it will have this) or you will need to define the panel reflectance in the array. If you exclude the panel capture and `useDLS=True` then the DLS data will be used if present. 

See the logic for this in [capture.create_alinged_capture](https://github.com/micasense/imageprocessing/blob/master/micasense/capture.py#L330)

If you find that this isn't what's happening, please let us know here.

esteps, 2019-05-16:
Thanks for the clarification! I see now how the logic works for the script.

poynting, 2019-05-16:
Sounds good, glad you got it working. Cheers.

closed 2019-05-16 by poynting