micasense/imageprocessing #64: Batch Processing and georeferencing

dunckelk, 2019-05-29:
The last step in the batch processing work flow is to write spatial metadata back to the images (.tif), however, my images just show up on top of each other in a GIS - when I check the metadata (i.e., log.csv) it has different coordinate data for each, why does this not translate to their positioning in a GIS?

import subprocess

old_dir = os.getcwd()
os.chdir(outputPath)
cmd = 'exiftool -csv="{}" -overwrite_original .'.format(fullCsvPath)
print(cmd)
try:
    subprocess.check_call(cmd)
finally:
    os.chdir(old_dir)

poynting, 2019-05-29:
That's not the purpose of that code. The original geotags are added back in
so that the stacked images can then be processed in a photogrammetry suite.
The orientations and locations of the images are not available without this
further processing.

dunckelk, 2019-05-29:
Hey, thanks Justin,

It would be much cooler if it was. Do you have any recommendations for where and how to georeference?

__________________________
Kathleen L Dunckel, PhD
Associate Professor
School of Environmental Citizenshp
Unity College
90 Quaker Hill Rd.
Unity, ME 04915
207.509.7127

poynting, 2019-05-29:
For most applications that's the purpose and function of a photogrammetry
suite like pix4d or agisoft.

Alternatively it would be possible to directly georeference the images if
you were to provide very accurate pose information during the flight or had
that information available post flight, but this usually requires a very
expensive RTK GPS with an accurate IMU to provide that information
separately.

The result is the same either way: it's not enough to know where the images
are taken at relatively low GPS precision, you that information much more
accurately and you also need to know what the (very accurate) pose
information is for each image. You also need the shape of the surface
you're imaging, which is also provided by the photogrammetry process.

All of this information has to come from an external source to be able to
do what you're looking for.

dunckelk, 2019-05-29:
Ok, can I use Atlas for that? Also (since you are responding to me, which is great!) does the Batch Processing code account for the basic radiometric corrections and conversion to reflectance, or do I have to do that separately?

Thank you, Justin!

Kathleen
__________________________
Kathleen L Dunckel, PhD
Associate Professor
School of Environmental Citizenshp
Unity College
90 Quaker Hill Rd.
Unity, ME 04915
207.509.7127

poynting, 2019-05-29:
The purpose of the batch processing code is to create an open reference implementation of our recommended radiometric calibration process, which can then be paired with a photogrammetry suite to generate geo-referenced maps.  It's not intended as a general production solution or a replacement for those processing options.

There are three primary users of this code in mind:

1. Researchers looking to publish that need an open and repeatable radiometric correction process (in contrast to the closed black box radiometric processes available in off the shelf tools).
2. Processing partners (Pix4D, Agisoft, SimActive, Solvi, Agribotics, etc. etc.) that need a reference radiometric implementation by which to validate or compare proprietary radiometric processing engines.
3. Anyone that wants to perform analysis on individual capture stacks without building maps.

Atlas does not process individual images.  If your use case is the more general "I took pictures and I want a map without much hassle," I recommend using one of the above named systems.

For more generalized support outside of specific issues with this codebase, it's best to use the standard MicaSense support process.

closed 2019-05-29 by poynting