micasense/imageprocessing #65: Plotutils issue ('numpy' has no attribute 'matrix')

Tossawon, 2019-06-05:
After i used plotutils library,
->plotutils.plotwithcolorbar(imageRaw, title='Raw image values with colorbar')
This error occurred.
~/miniconda3/envs/micasense/lib/python3.6/site-packages/numpy/core/fromnumeric.py in 
-> 1684     if isinstance(a, np.array):
   1685         return asarray(a).ravel(order=order)
   1686     else:

-->AttributeError: module 'numpy' has no attribute 'matrix'

No attribute maxtrix for numpy. This is the error from 'fromnumeric.py' or numpy? 
Should i reinstall numpy?

Tossawon, 2019-06-05:
I restarted the kernel, then everything work fine

closed 2019-06-05 by Tossawon

Jorgeagv98, 2022-01-12:
Man, I had the same error shutdown the kernel then works perfectly.

Thanks for the help