micasense/imageprocessing #66: Does pyzbar in Panel class work with Python3.6?

nickmorales, 2019-07-09:
Hi all, in the readme it says; however, pyzbar docs say it works with python 3. Does the auto calibration work with pyzbar in python3.6?

micasense.Panel
The Panel class is a helper class which can automatically extract panel information from MicaSense calibrated reflectance panels by finding the QR code within an image and using the QR Code location and orientation information to find the lambertian panel area. The class then allows extraction of statistics from the panel area such as mean raw values, mean radiance, standard deviation, and the number of saturated pixels in the panel region. The panel object can be included standalone, or used within the context of a Capture object.

Note: For the automatic panel QR code finding functions of the library to work, zbar and it's python bindings must be installed. We have made every effort to ensure this fails gracefully if zbar isn't available. Unfortunately zbar is only available using Python 2.7, not Python 3. If you're using Python 3.x, the code available in '/micasense/panel.py' shows how to find QR codes in images and to find the panel area from the QR location. We're currently looking for Python QR code finding options that work across platforms and Python versions, let us know if you have one that supports location!

poynting, 2019-07-09:
Yes, it does work now with 3.6. We should fix the docs.

closed 2019-07-09 by nickmorales

nickmorales, 2019-07-09:
Is there a way to extract the panel specific "panel calibration" numbers for each band from the auto calibration or does this need to be input by hand?

poynting, 2019-09-04:
@nickmorales sorry I didn't see this question until now. The panel class auto-extracts the panel information for newer (RP04+) panels, so you won't need to input it if you're using a newer panel.  If you're using an older panel (more than about 2 years old), the information isn't in the QR code and will need to be provided manually.

TClarke2, 2024-08-04:
Hi Poynting, Just following up on this. I noticed in this thread it was mentioned that pyzbar was working but I have been trying to use it for a project and it has not been functioning. I've tried following the code in '/micasense/panel.py' but have not been successful. Is there any insight you could provide regarding this.