micasense/imageprocessing #67: Cannot copy XMP tags

nickmorales, 2019-07-10:
I calibrated an image and want to save it with the uncalibrated image's EXIF and XMP tags. I am using the command line exiftool and when i try:
exiftool -tagsfromfile micasenseimage.tif -all:all newCalibratedImage.tiff
the EXIF tags are copied as expected, but the XMP tags are not.

if i do:
exiftool -xmp:bandname micasenseimage.tif
i see the band name XMP tag, but the copying is not working

poynting, 2019-07-10:
For a question like that, I would hit up the exiftool forums; I'm not an expert in exiftool at all.  If you're on Linux or Mac, I've found exiv2 can work better for this sort of operation.

However, copying all of the exif tags into a calibrated image may not be what you're after. If your goal is to process radiometrically calibrated single images with Agisoft or Pix4D, I would just copy the lat/lon/alt/pixel pitch/focal length like is done in the batch processing tutorial.  Any more and you risk those programs applying their own radiometric processing. Usually for Pix4D, for example, you'll likely still want to use the reflectance map generation process to generate a well averaged map, and if there are any radiometric tags present, it'll re-apply those.

TL;DR: my approach for this would be to follow through the process we used in the Batch Processing notebook, but instead of creating stacked tiffs and a CSV file of tags for them, create single tiffs and a CSV file for them, then use exiftool to add that info back in just like the batch processing does.

nickmorales, 2019-07-10:
thanks. i used a similar approach to your Batch Processing, but kept it all in memory instead

closed 2019-07-10 by nickmorales

aewindle110, 2020-12-07:
@nickmorales can you please explain how you kept all in memory during Batch Processing. I am trying to write metadata to my stacked tifs, but it doesn't seem to be working. I run Batch Process, extract metadata to log.csv, and then write the metadata to the stacked tifs. However, when I print all the metadata from a single stacked tif, the metadata that I wrote to it is not there. 

```
meta = metadata.Metadata("S:/Micasense_WQ/09_16_20/0000SET/000/flight_stacks/ZCIyHB3fAa4OkIDugKTw.tif", exiftoolPath=exiftoolPath)
meta.print_all()
```
```
SourceFile: S:/Micasense_WQ/09_16_20/0000SET/000/flight_stacks/ZCIyHB3fAa4OkIDugKTw.tif
ExifTool:ExifToolVersion: 12.02
File:FileName: ZCIyHB3fAa4OkIDugKTw.tif
File:Directory: S:/Micasense_WQ/09_16_20/0000SET/000/flight_stacks
File:FileSize: 6388369
File:FileModifyDate: 2020:12:07 13:47:22-05:00
File:FileAccessDate: 2020:12:07 13:47:21-05:00
File:FileCreateDate: 2020:12:07 13:47:21-05:00
File:FilePermissions: 666
File:FileType: TIFF
File:FileTypeExtension: TIF
File:MIMEType: image/tiff
File:ExifByteOrder: II
EXIF:ImageWidth: 1219
EXIF:ImageHeight: 923
EXIF:BitsPerSample: 16 16 16 16 16
EXIF:Compression: 8
EXIF:PhotometricInterpretation: 1
EXIF:StripOffsets: (Binary data 12082 bytes, use -b option to extract)
EXIF:SamplesPerPixel: 5
EXIF:RowsPerStrip: 3
EXIF:StripByteCounts: (Binary data 7699 bytes, use -b option to extract)
EXIF:PlanarConfiguration: 2
EXIF:Predictor: 1
EXIF:ExtraSamples: 0 0 0 0
EXIF:SampleFormat: 1 1 1 1 1
Composite:ImageSize: 1219 923
Composite:Megapixels: 1.12513
```7

But I am looking for the metadata such as lat, lon, focal length, etc. that the Batch Processing script seems to be extracting.