micasense/imageprocessing #68: pytest.py failed to open an image with exiftool

macimovic, 2019-07-24:
Hi, I am having trouble using exiftool to open any image from the dataset you provided. Tried debugging and the message that exiftool provides is that the format is bad. Here is the output from testing 

============================= test session starts ==============================
platform linux -- Python 3.6.8, pytest-4.3.1, py-1.8.0, pluggy-0.9.0
rootdir: /home/agremo/Documents/imageprocessing, inifile: pytest.ini
plugins: remotedata-0.3.1, openfiles-0.3.2, doctestplus-0.3.0, arraydiff-0.3
collected 116 items

tests/test_capture.py FFFFFFFFFFFFFFFFFFFFFFEEEEEEEFE                    [ 26%]
tests/test_dls.py .........E.......                                      [ 41%]
tests/test_image.py EEEEEEEEEEEEE.EEE                                    [ 56%]
tests/test_imageset.py FFF                                               [ 58%]
tests/test_metadata.py .FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF.FF.F              [ 89%]
tests/test_panel.py FFFFFFFFFFFF                                         [100%]

==================================== ERRORS ====================================
______________ ERROR at setup of test_altum_horizontal_irradiance ______________

non_panel_altum_file_list = ['data/ALTUM1SET/000/IMG_0008_3.tif', 'data/ALTUM1SET/000/IMG_0008_2.tif', 'data/ALTUM1SET/000/IMG_0008_6.tif', 'data/ALTUM1SET/000/IMG_0008_5.tif', 'data/ALTUM1SET/000/IMG_0008_4.tif', 'data/ALTUM1SET/000/IMG_0008_1.tif']

    @pytest.fixture()
    def non_panel_altum_capture(non_panel_altum_file_list):
>       imgs = [image.Image(fle) for fle in non_panel_altum_file_list]

tests/conftest.py:70: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
tests/conftest.py:70: in <listcomp>
    imgs = [image.Image(fle) for fle in non_panel_altum_file_list]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <micasense.image.Image object at 0x7f7aff124320>
image_path = 'data/ALTUM1SET/000/IMG_0008_3.tif', exiftool_obj = None

    def __init__(self, image_path, exiftool_obj=None):
        if not os.path.isfile(image_path):
            raise IOError("Provided path is not a file: {}".format(image_path))
        self.path = image_path
        self.meta = metadata.Metadata(self.path, exiftool_obj=exiftool_obj)
    
        if self.meta.band_name() is None:
>           raise ValueError("Provided file path does not have a band name: {}".format(image_path))
E           ValueError: Provided file path does not have a band name: data/ALTUM1SET/000/IMG_0008_3.tif

micasense/image.py:72: ValueError
_____________________ ERROR at setup of test_altum_panels ______________________

panel_altum_file_list = ['data/ALTUM1SET/000/IMG_0000_6.tif', 'data/ALTUM1SET/000/IMG_0000_4.tif', 'data/ALTUM1SET/000/IMG_0000_5.tif', 'data/ALTUM1SET/000/IMG_0000_3.tif', 'data/ALTUM1SET/000/IMG_0000_2.tif', 'data/ALTUM1SET/000/IMG_0000_1.tif']

    @pytest.fixture()
    def panel_altum_capture(panel_altum_file_list):
>       imgs = [image.Image(fle) for fle in panel_altum_file_list]

tests/conftest.py:61: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
tests/conftest.py:61: in <listcomp>
    imgs = [image.Image(fle) for fle in panel_altum_file_list]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <micasense.image.Image object at 0x7f7aff07be48>
image_path = 'data/ALTUM1SET/000/IMG_0000_6.tif', exiftool_obj = None

    def __init__(self, image_path, exiftool_obj=None):
        if not os.path.isfile(image_path):
            raise IOError("Provided path is not a file: {}".format(image_path))
        self.path = image_path
        self.meta = metadata.Metadata(self.path, exiftool_obj=exiftool_obj)
    
        if self.meta.band_name() is None:
>           raise ValueError("Provided file path does not have a band name: {}".format(image_path))
E           ValueError: Provided file path does not have a band name: data/ALTUM1SET/000/IMG_0000_6.tif
[... 2105 lines cut ...]
_______________________________ test_altum_lwir ________________________________

altum_lwir_image_name = 'data/ALTUM1SET/000/IMG_0000_6.tif'

    def test_altum_lwir(altum_lwir_image_name):
>       img = image.Image(altum_lwir_image_name)

tests/test_panel.py:139: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <micasense.image.Image object at 0x7f7aff0b42b0>
image_path = 'data/ALTUM1SET/000/IMG_0000_6.tif', exiftool_obj = None

    def __init__(self, image_path, exiftool_obj=None):
        if not os.path.isfile(image_path):
            raise IOError("Provided path is not a file: {}".format(image_path))
        self.path = image_path
        self.meta = metadata.Metadata(self.path, exiftool_obj=exiftool_obj)
    
        if self.meta.band_name() is None:
>           raise ValueError("Provided file path does not have a band name: {}".format(image_path))
E           ValueError: Provided file path does not have a band name: data/ALTUM1SET/000/IMG_0000_6.tif

micasense/image.py:72: ValueError
================ 71 failed, 20 passed, 25 error in 9.70 seconds ================

illcucuy, 2019-08-01:
This is about as far as I've gotten.

poynting, 2019-08-23:
Can you run the setup notebook and let us know what happens there? 

https://micasense.github.io/imageprocessing/MicaSense%20Image%20Processing%20Setup.html

poynting, 2019-08-23:
It's possible there is an issue with the `git-lfs` install and the actual files aren't there but instead just the LFS pointers to the files are.  I have seen this happen if, for example, I don't have git-lfs installed when I do the git checkout, and then install `git-lfs` later.  The easiest way to fix this is to just wipe the checkout directory, make sure you have `git-lfs` installed, and then re-do the checkout.

poynting, 2019-09-04:
@macimovic or @illcucuy can you verify if installing git-lfs and re-cloning the repo fixed the issue for you?  I just did a fresh install on windows where I first installed those two tools and then anaconda python, and was able to follow the setup notebook without issue.

poynting, 2019-10-04:
Closing due to inactivity & lack of response.

closed 2019-10-04 by poynting