micasense/imageprocessing #72: aligning nearby images without roads

anishau, 2019-09-04:
We're trying to align some images from 10 ft up with a rock or few in them - nothing very repetitive, but the alignment tools provided do not seem to work well for this. I've also tried with a bundle of grass. I can pretty easily get a good alignment by eye, so it seems theoretically doable. Do you have hypotheses around what modifications I could make that would improve alignment in this case?

poynting, 2019-09-04:
Please provide an example capture and I can take a look.

anishau, 2019-09-04:
Here you go, thanks! 

[test_grass.zip](https://github.com/micasense/imageprocessing/files/3575358/test_grass.zip)

[test_4rocks.zip](https://github.com/micasense/imageprocessing/files/3575350/test_4rocks.zip)

poynting, 2019-09-04:
By changing the `pyramid_levels` from `0` to `3` it works fine for both images.  Note that the thermal imagery isn't aligned using any kind of matching -- just using the factory calibration -- so it will be offset slightly, but it doesn't appear to impact the results much given the difference in pixel size.

[image]

[image]

[Alignment-Rocks.zip](https://github.com/micasense/imageprocessing/files/3575434/Alignment-Rocks.zip)
[Alignment-Grass.zip](https://github.com/micasense/imageprocessing/files/3575435/Alignment-Grass.zip)

closed 2019-09-04 by poynting

anishau, 2019-09-04:
ah, thank you so much! greatly appreciated the explanation as well.