micasense/imageprocessing #77: Lost exif data

Sweet1Chili, 2019-10-07:
Hi,
I had to distort and clip images of RedEdge M with different settings (in ArcGIS) to compare the radiance values pixelwise. But now, the function "raw_image_to_radiance" from the RedEdge Image Processing Tutorial 1 doesn't work anymore. Because the metadata was gone, I tried to copy it from the original image with ExiftoolGUI. 
In ExiftoolGUI I had the error: "Oversiezd IFD0 TileByteCounts".

But in Python with the code of the tutorial the error was again: 
radianceImage1, L, V, R = msutils.raw_image_to_radiance(meta1, imageRaw1)
  File "C:\Python37\lib\site-packages\micasense-0.0.1-py3.7.egg\micasense\utils.py", line 35, in raw_image_to_radiance
TypeError: cannot unpack non-iterable NoneType object

Maybe just several exif data has to be added, but I couldn't read out something like this in the tutorial.
Have anybody a suggestion to fix this? I' m still a beginner in programming...
Thank you!

poynting, 2019-10-10:
Can you just use the code in the batch processing tutorial? If you set useDLS = False and comment out the lines that reference the panel image and panel capture, the output should be in radiance.  You may need to scale something or change the output data type to floating point though inside the capture.save_capture_as_stack method.

https://micasense.github.io/imageprocessing/Batch%20Processing.html

poynting, 2019-11-20:
Closing due to lack of response. If there's a reproducible issue with the library code that you can document here please reopen.

ekochak, 2020-04-27:
Hi,

I am a bit baffled with the definition of radiance images in the 10-band batch processing script. If we set useDLS = False and comment out the lines about panel images, the output should be RAW files again. I could not find any function which converts raw images to radiance in the script. Am i missing something?

poynting, 2020-04-27:
It looks like that feature inadvertently fell out when I transitioned the batch processing to be multi-threaded, so the conversion defaults to DLS correction if there's no panel irradiance passed in.

Looks like we need to carry through `img_type` into the `capture.stave_stacks` method and then add it to the `save_params_list` inside `imageset.save_stacks` method, and finally use it (`img_type=params['img_type']` when we call `cap.create_alinged_capture` inside `imageutils.save_capture.`  

The last thing will be to update `capture.save_capture_as_stack` to allow for radiance images and in this case change the stack output type to floating point, since radiance images won't have a fixed output range relative to reflectance images and won't intuitively fit into 16b (a single raw image can fit in 16b, but a group of raw images converted to radiance, can have a much larger dynamic range.  For example one image may have gain 1 exposure 1ms, and another could have gain 8 exposure 10ms, and then changing 1 bit will represent a substantially different change in radiance.

Since this isn't really the

ekochak, 2020-04-29:
@poynting thanks for the quick response! I have also tried batch processing script with different sets of parameters (DLS and panel on and off) for RedEdge camera. When both parameters were off, instead of radiance, the output was DLS corrected reflectance.  I see the problem, but i am not that good at coding to be able to fix it now.

PR #148 (2021-05-24): imageset.py refactor
First cut at a imageset.py refactor, to better expose the high level code.

- Starts to fix #77. [This](https://github.com/micasense/imageprocessing/issues/77#issuecomment-620129409) comment.
- Refactor `save_capture()` to `ImageSet` class proper. It's not called from anywhere else so the logic seems better. Especially since `Capture` has these methods built in.
- Add return value to `dls_irradiance()`. I don't think this is used anywhere though.
- Add docstrings and PEP8 cleanup
- Add [tqdm](https://tqdm.github.io/) progess bar

TODO:
- Delete `image_from_file()`? Seems u [... cut]
closed 2021-05-24 by poynting