micasense/imageprocessing #81: TypeError: findTransformECC() missing required argument 'inputMask'

rrowlands, 2019-10-17:
I have very much been struggling with this error

TypeError: findTransformECC() missing required argument 'inputMask' (pos 6)

I have tracked it down to the wrong opencv version, as per this pull request:

opencv/opencv#13837

However when I do this command:

opencv_version

It returns:

3.4.7

so I am left somewhat confused. It was also mentioned in that pull request that perhaps my python is requiring optional arguments for some reason? However my python version is 3.7.3, and is downloaded/included via miniconda so I'm not sure how it could by the fault of my python. Reinstalling everything, as mentioned by @moghi005 did not work for me.

For anybody else with this same issue, I have managed to find a hack. Open up the imageutils.py file (in the micasense source). Search for : cv2.findTransformECC. Change the line to this:

cc, warp_matrix = cv2.findTransformECC(grad1, grad2, warp_matrix, warp_mode, criteria, None, 5)

This happens consistently on new installs of Ubuntu Bionic.

poynting, 2019-10-17:
I've run into that as well on a newer install.  It seems someone made a breaking change to the OpenCV API in what should just be a bugfix release.  Very annoying.  I've fixed this in master by pinning openCV to a version <3.4.7.  Incidentally this also fixes #75, which seems to have also broken in 3.4.7 (but works in 3.4.4). I haven't tracked down the real root cause of #75 yet.

I think this is a reasonable fix since this new API seems to be the new standard (though I dislike bodges like this). Please let me know if it works for you and I'll PR.
```
            try:
                cc, warp_matrix = cv2.findTransformECC(grad1, grad2, warp_matrix, warp_mode, criteria, inputMask=None, gaussFiltSize=1)
            except TypeError:
                cc, warp_matrix = cv2.findTransformECC(grad1, grad2, warp_matrix, warp_mode, criteria)
```

poynting, 2019-10-18:
Fixed in #82 via the code above and merged to `master`. If that doesn't fix it for you please comment here and re-open.

closed 2019-10-18 by poynting