micasense/imageprocessing #84: Micasense Requirements Are Not Clearly Defined

rrowlands, 2019-10-23:
These requirements should be clearly documented somewhere here in the Micasense documentation. These should include limitations in terms of Camera hardware (Rededge, Altum, etc) as well as limitations in terms of Camera firmware. From what I've gathered, here is what I see the requirements might be:

Altum (any version) or RedEdge-(3/M/MX) camera with firmware v2.1.0 or more recent

However I am honestly guessing here based on what I've gathered from the docs and software.

poynting, 2019-10-23:
I've added the supported firmware version numbers to the readme (and the outline for a workaround for older versions is below).

With respect to your last question, Altum has 6 bands, and is supported.  There is also a currently open issue #79 to handle 10-band data.

As far as other cameras, this is software is primarily developed by MicaSense employees as a service to their customers. While the license is intentionally permissive, there is no plan to support cameras from other manufacturers in this library.

For older RedEdge firmware versions <2.1.0 (June 2017) there is a possible workaround that requires the camera to be on-hand. Effectively, you'll need to devise a way to:

* Upgrade the camera firmware to the latest version
* Cycle power on the camera and capture an image with the new firmware
* Find and store the missing radiometric parameters from the image/band with the new firmware into the old firmware.  

I suspect the easiest way to accomplish this would be to use `exiv2` on a command line in linux or mac, but I haven't done it.

rrowlands, 2019-10-24:
Thanks again Justin!

closed 2019-10-24 by rrowlands