micasense/imageprocessing #85: Better Error Handling

rrowlands, 2019-10-23:
Specifically, when the software is run with very old RedEdge-M hardware, the 'Software' EXIF header is missing. When this header is missing, the firmware check does not run (passes silently) and a different error is thrown because it is missing some other header. Micasense needs to throw a proper error when the 'Software' header is missing. Additionally, it would be nice to devise a wider strategy to define: what are some expected errors that could occur, and how can we reliably handle and throw proper errors.

poynting, 2019-10-23:
At a glance, the only place in the code the `EXIF:Software` tag is used is in the `metadata.firmware_version()` call, which is only required by it's tests and the `metadata.horizontal_irradiance_valid()` method. I'm not sure what downstream fallout there is of the Software tag missing (`metadata.firmware_version()` should just return `None`).

The exception which is thrown for older firmware is in the constructor of the `Image` class, since older firmware lacks the required radiometric calibration tags.  https://github.com/micasense/imageprocessing/blob/master/micasense/image.py#L73

Currently, as soon as a user tries to construct a `micasense.Image` object with an `image_path` pointing to a file that was produced with firmware prior to 2.1.0, the following exception should be thrown:
```
ValueError('Library requires images taken with RedEdge-(3/M/MX) camera firmware v2.1.0 or later. ' +
            'Upgrade your camera firmware to at least version 2.1.0 to use this library with RedEdge-(3/M/MX) cameras.')
```

This should happen well before the constructor can call `horizontal_irradiance_valid.` If that method were called, the first check should return False, as v2.1.0 or less won't have a `XMP:HorizontalIrradiance` tag.

As it seems like you have a better grasp on catching the issue you're describing, so it would be good to be more specific. I welcome a PR with sample data and tests that implements what you're hoping to accomplish. 

There are currently other errors that do occur and raise exceptions, but they're not very complete.  What else did you have in mind?

rrowlands, 2019-10-23:
The exact stacktrace is as follows:

Traceback (most recent call last):
  File "batch.py", line 240, in <module>
    main()
  File "batch.py", line 31, in main
    panelCap = capture.Capture.from_filelist(panelNames)
  File "/home/rich/dev/projects/uasdm/git/imageprocessing/micasense/capture.py", line 94, in from_filelist
    images = [image.Image(fle) for fle in file_list]
  File "/home/rich/dev/projects/uasdm/git/imageprocessing/micasense/capture.py", line 94, in <listcomp>
    images = [image.Image(fle) for fle in file_list]
  File "/home/rich/dev/projects/uasdm/git/imageprocessing/micasense/image.py", line 77, in __init__
    self.utc_time = self.meta.utc_time()
  File "/home/rich/dev/projects/uasdm/git/imageprocessing/micasense/metadata.py", line 126, in utc_time
    utc_time = datetime.strptime(str_time, "%Y:%m:%d %H:%M:%S")
TypeError: strptime() argument 1 must be str, not None

poynting, 2019-10-23:
Probably -- if there's no valid UTC time there is going to be a lot of downstream issues, especially with things like loading an ImageSet (which is sorted by capture time) or trying to compute irradiance data (which requires a valid date/time).

Since that tag has been present in RedEdge firmware since late 2015, I'm not sure that we want to do a lot of work to make the library support that case.  It may be best to just catch the error loading that Image, or take a look at the other tags via exiftool to see if there's a tag that would make sense to fallback to.

rrowlands, 2019-10-24:
OK thanks for the help. I think what makes the most sense here is for me to add a validation stage at the beginning of my "batch.py" program where I will simply check for the existence of some basic headers, like 'Software' and a few GPS headers whereby I will enforce the requirements outlined in #84. The reason for this is it doesn't seem like there is a clear/logical single point of entry into your core code.

Here's a full list of headers I just pulled from my image (in case you find a use for it):

2 Interesting things to note:
1. There are no GPS coordinates
2. I just found the firmware version number, v1.0.1, under the 'SwVersion' header.

```
ExifTool Version Number         : 10.98
File Name                       : IMG_0047_5.resized.tif
Directory                       : <removed>
File Size                       : 607 kB
File Modification Date/Time     : 2019:10:21 19:51:04-06:00
File Access Date/Time           : 2019:10:22 10:54:15-06:00
File Inode Change Date/Time     : 2019:10:22 09:27:01-06:00
File Permissions                : rw-r--r--
File Type                       : TIFF
File Type Extension             : tif
MIME Type                       : image/tiff
Exif Byte Order                 : Little-endian (Intel, II)
Image Width                     : 640
Image Height                    : 480
Bits Per Sample                 : 16
Compression                     : Uncompressed
Photometric Interpretation      : BlackIsZero
Fill Order                      : Normal
Strip Offsets                   : 8
Orientation                     : Horizontal (normal)
Samples Per Pixel               : 1
Rows Per Strip                  : 480
Strip Byte Counts               : 614400
Planar Configuration            : Chunky
Page Number                     : 0 1
XMP Toolkit                     : XMP Core 4.4.0
About                           : Pix4D Camera Information
Rig Name                        : RedEdge-M
Band Name                       : Red edge
Central Wavelength              : 717
Wavelength FWHM                 : 10
Model Type                      : perspective
Principal Point                 : 2.3468,1.70714
Perspective Focal Length        : 5.4257818745756774
Perspective Focal Length Units  : mm
Perspective Distortion          : -0.098822106541715429, 0.13598578561374006, -0.013822321460663504, -0.00031230026443515785, 0.00020405716071584818
Vignetting Center               : 615.83942666656446, 481.35197138400309
Vignetting Polynomial           : -1.4492251415854857e-05, -2.8799647949945517e-07, -7.2694153001342486e-09, 2.8802906445909384e-11, -4.1017899693747142e-14, 1.9821058293467371e-17
Band Sensitivity                : 0.12857326785623385
Rig Camera Index                : 4
Yaw                             : -142.30760607424097
Pitch                           : 9.969832471076268
Roll                            : -4.6599217475637653
Irradiance Exposure Time        : 0.10100000351667404
Irradiance Gain                 : 16
Irradiance                      : 0.66285091638565063
Irradiance Yaw                  : -138.39874324682665
Irradiance Pitch                : 13.137648440827098
Irradiance Roll                 : -8.410058972899332
Boot Timestamp                  : 1680
Radiometric Calibration         : 0.00029616625732269277, 8.6107491061452425e-08, -6.5906891052254107e-06
Flight Id                       : <removed>
Capture Id                      : <removed>
Trigger Method                  : 6
Pressure Alt                    : 1584.8616943359375
Dark Row Value                  : 5341, 5321, 5441, 5453
Serial                          : DL06-1820009-SC
Sw Version                      : v1.0.1
Sensor Id                       : 4
Center Wavelength               : 717
Bandwidth                       : 10
Time Stamp                      : 1657093
Exposure                        : 0.10100000351667404
Gain                            : 16
Spectral Irradiance             : 0.66285091638565063
Raw Measurement                 : 877
Off Measurement                 : 2297
Image Size                      : 640x480
Megapixels                      : 0.307
```

closed 2019-10-24 by rrowlands

poynting, 2019-10-24:
Ok. It appears that's not an original RedEdge image. It's been resized,
which may also explain some of the missing  data depending on what program
resized it (but maybe not).

Without GPS and time information, there is not a lot you can do other than
basic panel-based reflectance compensation, but it may be possible to get
that with some mods, or maybe it exists in an external source that could be
used to inject it.

poynting, 2019-10-24:
On second thought, I do think that's likely the problem. The SW Version tag
is the dls software version. The dls was manufactured in 2018, so it's
likely that camera was shipped in 2018, in which case it would have been
shipped with SW > 2.1. Something doesn't seem right.

There are other core tags missing, as if someone copied the XMP data from
another file but not the actual EXIF data. I suggest asking the source if
they have the original data as it came off the camera.

rrowlands, 2019-10-24:
Well you are definitely right! Thank you so much!

On an unrelated note, do you know what would cause micasense to output just a completely black image?

rrowlands, 2019-10-25:
Any idea why the alignment on this image is so very poor? Adding the panel doesn't seem to help at all either.

[IMG_0001_.zip](https://github.com/micasense/imageprocessing/files/3769913/IMG_0001_.zip)

poynting, 2019-10-25:
Im not able to look at this moment, but try setting pyramid_levels to 3 or
4 instead of 0.

rrowlands, 2019-10-25:
HOLY COW that totally fixed it! Thanks. I'm trying to produce and automated solution here though, so I don't know, is there any way that I can know what the pyramid level will be based on the input? What exactly is this setting?

poynting, 2019-10-25:
If you're trying to develop an automated system, read at least the first
few blocks of this closely
https://micasense.github.io/imageprocessing/Alignment.html

Also see

https://github.com/micasense/imageprocessing/blob/8f150561a515274ae0c365e631019aa05d459203/micasense/imageutils.py#L73

Especially the link on line 93.

Also call align with debug=True and multiprocessing=False in the notebook,
it'll display all the pyramid info and the intermediates.

rrowlands, 2019-10-25:
Thanks again! Looks like you're already doing some automation in calculating that pyramid level. Setting it to None seems to work quite well.