micasense/imageprocessing #93: Create command-line utility to mimic the Batch Processing notebook

poynting, 2019-11-09:
Notebooks are nice, but a command-line tool for stacking would be handy

Should, at minimum, take
-folder of input images to stack
-folder to store the results
-explicit path of image to use for alignment
-explicit path of image containing a panel
-whether or not to generate preview images (e.g. RGB, CIR, etc)
-(options) which bands to use for the previews (e.g. 3,2,1 for RedEdge RGB)

Potential additions/improvements:
- Substitute alignment image with  a file specifying the warp matrices
- Substitute panel image with a file containing a time series of irradiance information
- Option to stack in parallel using n cores
- Include reading of a vector polygon file to define area of images to process
- HTML output report containing map of images like the current notebook

jonnyforestGIS, 2020-02-19:
@poynting I dig in in the subject and find this tutorial that can help.
https://towardsdatascience.com/how-to-write-python-command-line-interfaces-like-a-pro-f782450caf0d

moDanilevicz, 2020-05-01:
Hi, I have been working a bit on that because I have to process a few Altum images. It is quite initial and some stuff would have to change to be able to process RedEdge images, but I can share that if you think it may be useful :)

moDanilevicz, 2020-05-14:
Hi, I did a rough version of this to process a batch of Altum images. Would be happy to share the code if you think it would be help, although my coding skills are not spectacular. 
:)

JPI93, 2020-05-24:
> Hi, I did a rough version of this to process a batch of Altum images. Would be happy to share the code if you think it would be help, although my coding skills are not spectacular.
> :)

I'm working on writing something similar at the moment. It'd be amazingly helpful if you could share your code - if you don't mind :)

fischer5, 2020-05-25:
I would also be interested in this, if possible! I am working with Altum imagery too and it would be very helpful as I'm not a coding professional either. Cheers :)

poynting, 2020-05-26:
I'm happy to check out a PR or gist if anyone has code they would like to share.

moDanilevicz, 2020-05-26:
Sure, no problems.
I will take a look at it to check if it is still running and share :)

moDanilevicz, 2020-05-27:
The script is uploaded here https://github.com/moDanilevicz/multispectral_img

To run a test you have to use a few flags:
radio_calib_draft.py -i <path to image folder> -p <path to reference panel folder> -o <output file folder> -t <thumbnails? True/False> --dls <use it?True/False>

The code itself can be improved (and organized) a lot, but hope this can help  :)

JPI93, 2020-05-27:
@moDanilevicz thanks for linking your code :)

poynting, 2021-04-19:
Thanks for working on this all. I have merged @JPI93's code from https://github.com/micasense/imageprocessing/pull/114 into a branch (reconcile-cli-113) and we'll plan to work from there.