micasense/imageprocessing #96: Can't open Micasense using Jupyter Notebook

obgnim, 2019-11-24:
The installation followed the guide, but the following error was prompted when starting the environment:
Traceback (most recent call last):
  File "D:\anaconda3\envs\micasense\Scripts\jupyter-notebook-script.py", line 10, in <module>
    sys.exit(main())
  File "D:\anaconda3\envs\micasense\lib\site-packages\jupyter_core\application.py", line 268, in launch_instance
    return super(JupyterApp, cls).launch_instance(argv=argv, **kwargs)
  File "D:\anaconda3\envs\micasense\lib\site-packages\traitlets\config\application.py", line 663, in launch_instance
    app.initialize(argv)
  File "<D:\anaconda3\envs\micasense\lib\site-packages\decorator.py:decorator-gen-7>", line 2, in initialize
  File "D:\anaconda3\envs\micasense\lib\site-packages\traitlets\config\application.py", line 87, in catch_config_error
    return method(app, *args, **kwargs)
  File "D:\anaconda3\envs\micasense\lib\site-packages\notebook\notebookapp.py", line 1679, in initialize
    self.init_webapp()
  File "D:\anaconda3\envs\micasense\lib\site-packages\notebook\notebookapp.py", line 1442, in init_webapp
    self.http_server.listen(port, self.ip)
  File "D:\anaconda3\envs\micasense\lib\site-packages\tornado\tcpserver.py", line 152, in listen
    self.add_sockets(sockets)
  File "D:\anaconda3\envs\micasense\lib\site-packages\tornado\tcpserver.py", line 165, in add_sockets
    self._handlers[sock.fileno()] = add_accept_handler(
  File "D:\anaconda3\envs\micasense\lib\site-packages\tornado\netutil.py", line 279, in add_accept_handler
    io_loop.add_handler(sock, accept_handler, IOLoop.READ)
  File "D:\anaconda3\envs\micasense\lib\site-packages\tornado\platform\asyncio.py", line 99, in add_handler
    self.asyncio_loop.add_reader(fd, self._handle_events, fd, IOLoop.READ)
  File "D:\anaconda3\envs\micasense\lib\asyncio\events.py", line 501, in add_reader
    raise NotImplementedError
NotImplementedError

poynting, 2019-11-24:
That appears to be a jupyter issue if jupyter isn't starting. I haven't
seen that before. It doesn't appear to be related to any of the code in
this repository.

poynting, 2019-11-24:
It appears that some change in python 3.8 broke tornado (one of the packages jupyter uses) according to https://github.com/tornadoweb/tornado/issues/2608

I just changed the micasense_conda_env.yml to pin python to 3.7, which will hopefully fix this until they get the upstream issues sorted out.

To downgrade to python 3.7, it will be necessary to need to delete the micasense conda environment,
```
conda remove --name myenv --all
```
then run 
```
git pull origin master
```
 in your anaconda window, and then re-create the micasense environment.
```
conda env create -f micasense_conda_env.yml
```

obgnim, 2019-11-24:
Thank you very much! I will try it.

obgnim, 2019-11-24:
The method you provided worked verywell. I have rebuilded the environment. But there is a new problem.
When i try to runnig the sample program provided in the web page: "MicaSense Image Processing Setup". The step "Testing image reading and panel detection", Error message like this:

ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-4-22d275e27f9f> in <module>
----> 1 from micasense.image import Image
      2 imagePath = os.path.join('.','data','0000SET','000')
      3 imageName = glob.glob(os.path.join(imagePath,'IMG_0000_1.tif'))[0]
      4 
      5 img = Image(imageName)

ModuleNotFoundError: No module named 'micasense'

poynting, 2019-11-24:
Are you running `jupyter notebook .` from the imageprocessing directory
that contains the micasense folder?

What happens when you run `pytest`?

obgnim, 2019-11-25:
I completely uninstalled the Micasense environment and deleted the folders，then reinstalled the Micasense environment with Git clone and Python commands. Everything is all right now. Thank you for your help.

closed 2019-11-25 by obgnim