micasense/imageprocessing #97: [Question] Create a QGIS Processing Plugin

jonnyforestGIS, 2019-12-06:
Hi guys,
recently my Institute acquires a Micasense Altum. I'm thinking to create a Processing plugin for QGIS with:

**Inputs:**
- Files or Directory
- Need more?

**Plugin features:**
1. Batch Import bands;
2. Align Bands;
3. Create stack bands in an image;
4. Mosaicking to a final product (optional) using Tool Mosaicking (SAGA GIS) or Leave these step* and use webODM [possabilitiy to have more products from the image] );

**Outputs:**
Stack images with all bands for a single image (or all area - depends from the decision of the 4th topic);

**So my question is:**
The tutorials in Jupyterbook can answer to the first 3 topics?

*For the 4th topic, I'm looking to an algorithm from SAGA Mosaicking. This can use single bands or a grid collection (collect of bands):

- [SAGA Tool Mosaicking](http://www.saga-gis.org/saga_tool_doc/7.4.0/grid_tools_3.html)

As an alternative, I was thinking of using webODM to stitch the images because of number o f derivated products (DTM, DSM, point cloud, etc).

What are your thoughts?

Kind regards,
João Gaspar

poynting, 2019-12-09:
A QGIS plugin might be nice; given the amount of peripheral QT stuff that gets dragged in it's probably best to do it in a separate repository and just use this library as an import.  I think the first step though would still be to work through https://github.com/micasense/imageprocessing/issues/93 which could be used to do things like easy creation of an imageset geojson which could then be loaded into QGIS.

As far as the 4th topic, the SAGA mosaicing tool looks more like the gdalmerge command, which just merges multiple geotiffs into a single geotiff.  In order to do that, you would still require a photogrammetry or stitching solution to create the georeferenced geotiffs from the stacked files, and most of those tools already output a merged geotiff.

WebODM is probably a more practical route; see the active development in e.g. https://github.com/OpenDroneMap/WebODM/issues/297

rdehaan1971, 2020-12-30:
If you can create the plugin to do the raw to reflectance calibration for a single file Altum 6 band image by allowing it to point to a folder containing the altum panel and raw images you would solve my problems :)