micasense/imageprocessing #98: **'reflectance_from_panel_serial' is not defined.**

tejasri19, 2019-12-23:
'Panel' object has no attribute 'reflectance_from_panel_serial' & ype object 'Capture' has no attribute 'reflectance_from_panel_serial'.

poynting, 2019-12-23:
1. Have you run the setup notebook successfully?
2. Can you run the tests?
`pytest .`
3. If 1 and 2 are successful, please provide a full trace of what code is
failing and the full log message.

tejasri19, 2019-12-23:
> 1. Have you run the setup notebook successfully? 
Yes, I could run the setup successfully.
> 2. Can you run the tests? `pytest .`
some tests are failed and some are passed.

tejasri19, 2020-01-07:
@poynting 
I was able to run the setup successfully. When I run 'pytest' , I'm getting several errors.  Please helP

`>           raise ValueError("Provided file path does not have a band name: {}".format(image_path))
E           ValueError: Provided file path does not have a band name: data/ALTUM1SET/000/IMG_0008_5.tif

micasense/image.py:73: ValueError
___________________________________________________________________________________________ test_altum_from_filelist ___________________________________________________________________________________________

non_panel_altum_file_list = ['data/ALTUM1SET/000/IMG_0008_5.tif', 'data/ALTUM1SET/000/IMG_0008_1.tif', 'data/ALTUM1SET/000/IMG_0008_4.tif', 'data/ALTUM1SET/000/IMG_0008_3.tif', 'data/ALTUM1SET/000/IMG_0008_6.tif', 'data/ALTUM1SET/000/IMG_0008_2.tif']

    def test_altum_from_filelist(non_panel_altum_file_list):
>       cap = capture.Capture.from_filelist(non_panel_altum_file_list)

tests/test_capture.py:178: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
micasense/capture.py:97: in from_filelist
    images = [image.Image(fle) for fle in file_list]
micasense/capture.py:97: in <listcomp>
    images = [image.Image(fle) for fle in file_list]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <micasense.image.Image object at 0x7f072ad1f0b8>, image_path = 'data/ALTUM1SET/000/IMG_0008_5.tif', exiftool_obj = None

    def __init__(self, image_path, exiftool_obj=None):
        #print(image_path)
        if not os.path.isfile(image_path):
            raise IOError("Provided path is not a file: {}".format(imanon_panel_rededge_file_listge_path))
        self.path = image_path
        self.meta = metadata.Metadata(self.path, exiftool_obj=exiftool_obj)
    
        if self.meta.band_name() is None:
>           raise ValueError("Provided file path does not have a band name: {}".format(image_path))
E           ValueError: Provided file path does not have a band name: data/ALTUM1SET/000/IMG_0008_5.tif

micasense/image.py:73: ValueError
_________________________________________________________________________________________ test_altum_from_single_file __________________________________________________________________________________________

altum_flight_image_name = 'data/ALTUM1SET/000/IMG_0008_1.tif'

    def test_altum_from_single_file(altum_flight_image_name):
>       cap = capture.Capture.from_file(altum_flight_image_name)

tests/test_capture.py:183: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
micasense/capture.py:88: in from_file
    return cls(image.Image(file_name))
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <micasense.image.Image object at 0x7f072ab48898>, image_path = 'data/ALTUM1SET/000/IMG_0008_1.tif', exiftool_obj = None

    def __init__(self, image_path, exiftool_obj=None):
        #print(image_path)
        if not os.path.isfile(image_path):
            raise IOError("Provided path is not a file: {}".format(imanon_panel_rededge_file_listge_path))
        self.path = image_path
        self.meta = metadata.Metadata(self.path, exiftool_obj=exiftool_obj)
    
        if self.meta.band_name() is None:
>           raise ValueError("Provided file path does not have a band name: {}".format(image_path))
E           ValueError: Provided file path does not have a band name: data/ALTUM1SET/000/IMG_0008_1.tif

micasense/image.py:73: ValueError
____________________________________________________________________________________________ test_no_rig_relatives _____________________________________________________________________________________________

non_panel_rededge_file_list = ['data/0000SET/000/IMG_0001_2.tif', 'data/0000SET/000/IMG_0001_5.tif', 'data/0000SET/000/IMG_0001_4.tif', 'data/0000SET/000/IMG_0001_3.tif', 'data/0000SET/000/IMG_0001_1.tif']

    def test_no_rig_relatives(non_panel_rededge_file_list):
>       cap = capture.Capture.from_filelist(non_panel_rededge_file_list)

[... 952 lines cut ...]

flight_image_name = 'data/0000SET/000/IMG_0001_1.tif'

    def test_panel_not_detected(flight_image_name):
>       img = image.Image(flight_image_name)

tests/test_panel.py:118: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <micasense.image.Image object at 0x7f072ace5dd8>, image_path = 'data/0000SET/000/IMG_0001_1.tif', exiftool_obj = None

    def __init__(self, image_path, exiftool_obj=None):
        #print(image_path)
        if not os.path.isfile(image_path):
            raise IOError("Provided path is not a file: {}".format(imanon_panel_rededge_file_listge_path))
        self.path = image_path
        self.meta = metadata.Metadata(self.path, exiftool_obj=exiftool_obj)
    
        if self.meta.band_name() is None:
[... cut ...] ise ValueError("Provided file path does not have a band name: {}".format(image_path))
E           ValueError: Provided file path does not have a band name: data/0000SET/000/IMG_0001_1.tif

micasense/image.py:73: ValueError
================================================================================== 67 failed, 19 passed, 41 errors in 22.88s ===================================================================================
(micasense) tejasrii@winet:~/Image-ExifTool-10.98/imageprocessing$ `

poynting, 2020-01-07:
Most likey you did not have git-lfs installed when cloning the repository,
or the git lfs fetch failed for some reason, so the example files are not
actually files but just lfs pointers.

To test run exiftool.exe on one of the files from a command prompt.

https://github.com/micasense/imageprocessing/blob/master/README.md#how-do-i-get-set-up

tejasri19, 2020-04-24:
> Most likey you did not have git-lfs installed when cloning the repository, or the git lfs fetch failed for some reason, so the example files are not actually files but just lfs pointers. To test run [...]

Thanks a lot! :)

closed 2020-04-24 by tejasri19