micasense/imageprocessing #99: panels

obgnim, 2020-01-06:
Why can't my panel image display?I have imported the panel images, but it reminds me that I cannot find the panel images.My path is correct, the old images are ok, but the new images are not.

poynting, 2020-01-06:
1. Run the setup notebook and the tests. Make sure to fix any errors that appear.
2. If the problem persists, provide a very simple example code that shows the problem, and any image files that cause it to fail.

obgnim, 2020-01-07:
Thanks for your reply. I have solved the problem. The reason why I can't recognize it is that the panel image has been compressed.But I still want to know, if there are shadows on the QR code,  can we not recognize this QR code?

obgnim, 2020-01-07:
Thanks for your reply. I have solved this problem because the picture is compressed.But I still want to know whether the qr code of this image cannot be recognized if it is partially covered by shadows.

poynting, 2020-01-07:
As a general point, QR codes partially covered by shadows can be difficult for software to recognize.

closed 2020-01-10 by poynting