micasense/imageprocessing #100: Converting images to reflectance

sadouglas, 2020-01-08:
Hello, I am using tutorial 1 to convert images to reflectance using the calibration panel. Under the "In Practice" section, a flight image is converted to reflectance. Since meta is not redefined after the calibration panel, is the metadata used in this section from the calibration panel and not from the actual flight image?

poynting, 2020-01-09:
Yes, you are correct. The metadata in the example is for the calibration image, but it should be loaded separately from the flight image for the conversion.

```
flightImageName = os.path.join(imagePath,'IMG_0001_4.tif')
flightImageRaw=plt.imread(flightImageName)
plotutils.plotwithcolorbar(flightImageRaw, 'Raw Image')

flightImageMeta = metadata.Metadata(flightImageName, exiftoolPath=exiftoolPath)
flightRadianceImage, _, _, _ = msutils.raw_image_to_radiance(flightImageMeta, flightImageRaw)
flightReflectanceImage = flightRadianceImage * radianceToReflectance
flightUndistortedReflectance = msutils.correct_lens_distortion(flightImageMeta, flightReflectanceImage)
plotutils.plotwithcolorbar(flightUndistortedReflectance, 'Reflectance converted and undistorted image');
```

Unless you're trying to re-create the calibration process, I would generally recommend using the `Image` and `Capture` classes for the conversions.  For example, using a recent panel and flight image, it's possible to generate reflectance images following the process in the first cell of the Alignment tutorial. If using an older panel, it will be necessary to set panel_albedo to the band-ordered list of panel albedos for your panel capture.

Capture Example:
```
import os, glob
import micasense.capture as capture
%matplotlib inline

imagePath = os.path.join('.','data','ALTUM1SET','000')
imageNames = glob.glob(os.path.join(imagePath,'IMG_0245_*.tif'))
panelNames = glob.glob(os.path.join(imagePath,'IMG_0000_*.tif'))

panelCap = capture.Capture.from_filelist(panelNames)
flightCap = capture.Capture.from_filelist(imageNames)

#if your panel doesn't have albedo info built in, use a list, e.g [0.52,0.55,...]
panel_albedo = panelCap.panel_albedo() 
panel_irradiance = panelCap.panel_irradiance(panel_albedo)
flightCap.plot_undistorted_reflectance(panel_irradiance)
```

It's also possible to do this on single images instead of captures, but it's a little more complicated because Image is missing some of the helper functions in `Capture`.  

Image Example:
```
import os, glob
import micasense.image as image
import micasense.panel as panel
%matplotlib inline

imagePath = os.path.join('.','data','0000SET','000')
imageName = os.path.join(imagePath,'IMG_0001_4.tif')
panelName = os.path.join(imagePath,'IMG_0000_4.tif')

panelImg = image.Image(panelName)
flightImg = image.Image(imageName)

# Use the panel object to extract panel information (if available)
p = panel.Panel(panelImg)
panel_albedo = 0.61 # or p.reflectance_from_panel_serial()
panel_irradiance = p.irradiance_mean(panel_albedo)

#no Image corollary to plot_undistorted_reflectance, so we can call plotutils directly
import micasense.plotutils as plotutils
title = '{} Band {} Reflectance'.format(flightImg.band_name, flightImg.band_index)
undistorted_reflectance_img = flightImg.undistorted(flightImg.reflectance(panel_irradiance))
plotutils.plotwithcolorbar(undistorted_reflectance_img, title=title)
```

sadouglas, 2020-01-09:
Thanks for the info, very helpful!
However, when I try to run your capture example, I receive the following error:
ValueError: Provided file path does not have a band name: ./data/ALTUM1SET/000/IMG_0000_4.tif

poynting, 2020-01-09:
Most likely git-lfs was not installed when cloning the repository, or the git lfs fetch failed for some reason, so the example files are not actually files but just lfs pointers.  When this happens I find it easiest to just wipe out and re-clone the repository. If you go this route be mindful to copy any files that have changed that you need to keep.

https://github.com/micasense/imageprocessing/blob/master/README.md#how-do-i-get-set-up

poynting, 2021-04-19:
Closing due to inactivity. Please let us know if there is still an issue to resolve.

closed 2021-04-19 by poynting