micasense/imageprocessing #101: Alignment notebook throws error

gemcosta, 2020-01-09:
Using the `micasense` conda environment, and launching the Alignment notebook to stack a single capture from a RedEdge-3 on v3.4.3 is throwing an error on Step 1.1 (Opening Images):

` TypeError: unsupported operand type(s) for /: 'float' and 'NoneType'`

At first, I believed this to be possibly because a RedEdge-3 on an older firmware does not have Rig Relatives. However, I get the same error when running the code on an Altum on v1.3.6 with rig relatives. So it appears this error is not related to the camera or data being imported.

This error persists on multiple machines, which leads me to believe there is something in the code that needs to be updated.

poynting, 2020-01-09:
Can you post the whole error here?  My quick guess is that the panel albedo can't be derived from the panel images and will need to be set manually, possibly because an older panel is being used.

The handling of this type of case was improved in the newer 10-band alignment notebook; it may be worth copying that code over and seeing if you get a different result.

gemcosta, 2020-01-09:
Sure, below is the full error. Sorry that the formatting doesn't translate very well.

```
→---------------------------------------------------------------------------
TypeError Traceback (most recent call␣
,→last)
<ipython-input-2-c6820fa14813> in <module>
39 else:
40 panel_reflectance_by_band = [0.67, 0.69, 0.68, 0.61, 0.67]␣
,→#RedEdge band_index order
---> 41 panel_irradiance = panelCap.
,→panel_irradiance(panel_reflectance_by_band)
42 img_type = "reflectance"
43 capture.plot_undistorted_reflectance(panel_irradiance)
~/imageprocessing/micasense/capture.py in panel_irradiance(self,␣
,→reflectances)
286 irradiance_list = []
287 for i,p in enumerate(self.panels):
--> 288 mean_irr = p.irradiance_mean(reflectances[i])
289 irradiance_list.append(mean_irr)
290 return irradiance_list
~/imageprocessing/micasense/panel.py in irradiance_mean(self,␣
,→reflectance)
234 def irradiance_mean(self, reflectance):
235 radiance_mean, _, _, _ = self.radiance()
--> 236 return radiance_mean * math.pi / reflectance
237
238 def plot_image(self):
TypeError: unsupported operand type(s) for /: 'float' and 'NoneType'

```

poynting, 2020-01-09:
I see the issue, for a case like this where the panel QR does not have embedded albedo information:
```
if panelCap is not None:
    if panelCap.panel_albedo() is not None:
```
should be 
```
if panelCap is not None:
    if not None in panelCap.panel_albedo():
```

poynting, 2020-01-10:
I update the library to better handle the case above. This should now be fixed by #102.  Try a 
```
git pull origin master
```
within your imageprocessing directory to get the changes and let us know if that works for you.

For the case where the panel is older (serial number starts with RP01, RP02, or RP03), it will be necessary to manually enter the panel albedo as a list in the first cell of the alignment notebook, e.g. 
```
panel_reflectance_by_band = [0.50, 0.51, 0.52, 0.53, 0.54]
```

closed 2020-03-10 by poynting